wills106/homeassistant-solax-modbus #1264: Spring clean Bounty Hunt 💐

wills106, 2025-02-23:
### Description

At the time of writing there are 128 Open Issues.
I am struggling to keep track and process all of them, some of the issues are potentially `Operator Error` (Controversial I know...)
Some I can't replicate.
Some could be very stale, like a mouldy French Fry down the side of your car seat 🤔

### Suggested Solution

Unfortunately there wont be a financial award, but I would be grateful if either you could do the following:
- Close down Stale Issues
- Report Issues that are potentially fixed, either in the actual issue or here
- Read https://homeassistant-solax-modbus.readthedocs.io/en/latest/ before posting a new Issue
- Searching for a duplicate Issue

wills106, 2025-02-23:
Ps, please don't use this as a rant for why XYZ isn't fixed or why XYZ isn't implemented ❤

w4iq3, 2025-02-25:
I would like to (try to) help!

MehrazRumman, 2025-02-25:
I wan to contribute !

viking2010, 2025-04-02:
@wills106 I've been through a few of the Solis tickets (as that's the brand I have) and questioned a few people on whether their issues are now resolved. There's quite a few close to or over a year old. I'd be tempted to close tickets off if there have been no updates or responses to requests for more information in the last couple of months. That should help prune some of the tickets down for you 👍

wills106, 2025-04-02:
Yea I should look into one of those Stale bots. Then if no one responds to it closing down warning, the issue can be automatically closed down.

closed 2025-09-16 by github-actions[bot]