wills106/homeassistant-solax-modbus #1265: [Bug]: 'Growatt' - 'MOD10KTL3-XH' ErrorLog: value has state class total_increasing, but its state is not strictly increasing

derjape, 2025-02-25:
### Describe the bug

I got daily entries in my HA Log , that **_Entity sensor.growattwr_today_s_yield from integration solax_modbus has state class total_increasing, but its state is not strictly increasing._**  also **_Entity sensor.growattwr_total_yield_**.
See full Log further down.
Don't know if it is leading to any failures but wanted to report it here.

### Integration Version

2025.02.1

### Homeassistant core version

2025.2.5

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

DN1.0ZBDB

### Connection Method

Waveshare RS485 to ETH (B)

### Detailed Error Log

Logger: homeassistant.components.sensor.recorder
Quelle: components/sensor/recorder.py:317
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 10:25:10 (2 Vorkommnisse)
Zuletzt protokolliert: 10:25:10

Entity sensor.growattwr_today_s_yield from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 5.5 (5.6) with last_updated set to 2025-02-25T09:24:58.930116+00:00. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Entity sensor.growattwr_total_yield from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 9927.1 (9927.2) with last_updated set to 2025-02-25T09:24:58.930155+00:00. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

PSMGoossens, 2025-03-09:
@derjape  Are you completely sure? Did you have a hickup with data connectivity? Because I'm doubting that yield today could be lowered :)
Could you please provide the screenshot of the data?

derjape, 2025-03-09:
I provide you any data you need for bugfinding.
I hope the following Screenshots will help. Otherwise ket me know, what exactly you need.

[image]

[image]

[image]

[image]

PSMGoossens, 2025-03-09:
@derjape Thank you for the  provided information and indeed I'm seeming that the values are sometimes lowering, which I don't understand. I've checked it with my converter if it has the  same issue ( MOD 5000 TL3-XH), but in my case it's always increasing. 
Could it be a firmware issue of the  Growatt inverter? Did you ever request a new firmware version for your inverter from Growatt?

If that is the case we have to dive deeping in why the yield could be dipping sometime (could be double rounding maybe?), but lets first check the firmware

rmk-sys, 2025-03-13:
hi, changing state class to 'TOTAL' in total_yield and today's_yield is solving the problem!

closed 2025-09-16 by github-actions[bot]