wills106/homeassistant-solax-modbus #1270: suddenly all Solax entities/sensors are unavailable

Hubby33, 2025-03-05:
Hi there,
as of this morning, all my 244 solax entities are unavailable. I already tried to reload the Solax Inverter Modbus, which did not change anything.
Then I completely uninstalled and tried to re-install it. Here it hangs in the configuration (the second window, where you need to enter IP and port of the stick )
Was there any change to port, maybe?
My Solax Cloud, by the way, still is up and running....

Thanks
Hubby

Hubby33, 2025-03-05:
never mind on that. Rebooting my router did the trick.....

sumitkm, 2025-03-06:
@Hubby33 unplugging and replugging the wifi dongle on the inverter did it for me :). 

Please close this ticket if you are no longer need it investigated.

closed 2025-03-07 by wills106