wills106/homeassistant-solax-modbus #1275: [Bug]: Integration of X1 Mini Inverter don´t work after each Update of HomeAssistant Core or even SolaX-Modbus

edli83, 2025-03-09:
### Describe the bug

I have 3 Inverters in my System.
1 SolaX X3 G4 Hybrid and 3 X1 Mini per phase due to 22 Modules for the X3 and 3 x 5 Modules for each X1.
The X3 is connected via WiFi/ LAN Dongle and works fine.
The 3 X1 are connected via 1 Waveshare RS485 to Eth, separated with differnt Modbus adresses.

Problem:
After each update, not depending whether it´s an update of HomeAssistant or SolaX Modbus Integration, I have big trouble to get the X1 working again.
Mostly it helps to delete the X1 in the integration, restart HomeAssistant, reinstall the X1 with same name - then they worked again till last SolaX Modbus Integration to Ver 2025.02.1b3.
Now after each night (the Inverters shut down over night due to no PV Power) the integration do not work and I have no Sensors/ entities filled with data.
This should not be the right way. All inverters should work again after any update. This "reload" of all inverters is lost time.
Thank you for help - hopefully it is a simple fault.
If any additional information would be necessary, please don´t hesitate and ask. I will try to deliver as soon as possible.

### Integration Version

2025.02.1b3

### Homeassistant core version

2025.3.1

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H34A10 and 3 x XM3302

### Inverter firmware versions

X3:  ARM 1.47-1.09 - DSP 1.46 |  X1 ARM 1.52 - DSP 2.35

### Connection Method

X3: WiFi/ LAN Dongle  | X1: Waveshare RS485 to Eth

### Dongle firmware

X3: 1.005.04  |  X1: 3.006.04

### Detailed Error Log

[home-assistant_2025-03-09T11-00-42.185Z.log](https://github.com/user-attachments/files/19149118/home-assistant_2025-03-09T11-00-42.185Z.log)

### Additional context

[image]
[image]

edli83, 2025-03-09:
Addition/ Comment:
With a fewer version of SolaX Modbus Integration also the X1 worked good, even after updates.
Unfortunately I´m not sure with which version this was - sorry.

PSMGoossens, 2025-03-09:
@edli83 What did you configure as timeout settings in the Eth to modbus converter? I think it could be a timeout issue on the converter which doesn't allow a new connection. Could you try to lower it? [Some docs](https://www.manualslib.com/manual/1692277/Waveshare-Rs485-To-Eth.html?page=20)

edli83, 2025-03-09:
See current waveshare settings added screenshot.
Is this then in the line Advanced Settings?
What should I try to set.

Thank you

[image]

edli83, 2025-03-09:
by the way, my waveshare looks like this:

[image]

edli83, 2025-03-09:
I now modified the "Enable Multi-Host" Setting to "yes".
This seems to help. After this new setting, all X1 inverter very fast appears with data.
Could this be the wrong setting?

PSMGoossens, 2025-03-09:
> I now modified the "Enable Multi-Host" Setting to "yes".
Multi-host would help, since the not timeout connection will stay open and a new one could be made

For the Wireshare you could donwload the vircom software and also edit the timeout

[image]
you could try to lower it to +/- 25 seconds and try it again. or leave mutlhost for now :)

edli83, 2025-03-09:
Vircom I already use.
I lower Keep Alive Time now to 25.
[image]

Multi-Host still is now active.
[image]

Seems to work
Anything else to check?

Thank you for your very quick help.

PSMGoossens, 2025-03-09:
Try to reboot HomeAssistant a few times. If you don't have any problems anymore, then I think you could close the issue 😄

edli83, 2025-03-09:
I already have rebooted a few times and everything seems to work as it shoud.
Many thanks to your really quick help.

This written - I will colose the issue.

closed 2025-03-09 by edli83

edli83, 2025-03-12:
Only to underline after current update of HomeAssistant Core:
Immedeately after restart after the update all inverters (incl. the X1) work's fine.

Once more many thank's for your super help.