wills106/homeassistant-solax-modbus #1281: [FR]: 'Solax X3-MGA-50k-G2 smartmeter informations

msporik, 2025-03-14:
### Description

Hi, I have at my house HA server connected with Solax G4 10K inverter, and in HA I have over 150 entities.
Now, I have tried by my uncles house HA connected with Solax X3-MGA-50k-G2 inverter. Connection succesfull, but I see in HA only 41 entities. For me, it seems I dont see informations from smartmeter, which is correctly connected to inverter.
Any idea what can I try ?

### Suggested Solution

any idea what can I try ?

wills106, 2025-03-15:
There are hardly any registers for the Mega / Forth.
Unless there is a new protocol document available there isn't anything else to add of any real use.

wills106, 2025-03-15:
I have marked this as `wontfix` for now unless anyone can provide newer documentation.

closed 2025-03-15 by wills106

msporik, 2025-03-15:
Thanks a lot.
We will see, if something useful come