wills106/homeassistant-solax-modbus #1284: [Bug]: Solax G4 hybrid Inverter Work Mode can't switch to Manual.

Bl00d-B0b, 2025-03-14:
### Describe the bug

Hello,

X3-Hybrid by SolaX Power
Firmware: DSP v1.41 ARM v1.40
Hardware: Gen4

When try to switch "Inverter Work Mode" to manual, then I receive status Unknown!
X3-15K-G4 Inverter Work Mode became unknown
16:50:06 - 1 minute ago
X3-15K-G4 Inverter Work Mode changed to Manual triggered by action Select: Select
16:49:59 - 2 minutes ago

### Integration Version

2025.03.2

### Homeassistant core version

2025.3.2

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

ARM Version 1.40-1.15  Main DSP Version 1.41 

### Connection Method

Lan 3.0

### Dongle firmware

LAN:   3.005.02

Bl00d-B0b, 2025-03-14:
Modes "Disabled" and "Smart Save" can be switched

steps56, 2025-03-14:
Did you switch to "unlock advanced" before trying anny write settings?
Without this state, no way to send those write commands.

Bl00d-B0b, 2025-03-15:
Yes. I try both modes. Unlocked and unlocked advanced. As I mentioned I can switch to disabled and smart save. But when I set manually it's changes mode to unknown.

wills106, 2025-03-15:
It probably is switching to Manual Mode, but it's going unknown as there is a typo in the read register part.

wills106, 2025-03-15:
Just released [2025.03.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.03.4)

Bl00d-B0b, 2025-03-15:
Works

Bl00d-B0b, 2025-03-15:
Fixed

closed 2025-03-15 by Bl00d-B0b