wills106/homeassistant-solax-modbus #1285: [FR]: Solax Gen2 EVC X1|X3-HAC-* support

jefft4, 2025-03-15:
### Description

Add support for the Solax Gen2 EVC range.

### Suggested Solution

I have a Gen2 X1-HAC-7 and I've got the Solax modbus documentation for the Gen2. Looks mostly the same sa Gen1, some extra values and possibly a couple of register changes. 

I'll clone the repo and generate a pull request for the changes I can identify.

Will need someone with a X3-HAC-* to identify the serial number range for those - add later.

jefft4, 2025-03-15:
https://github.com/wills106/homeassistant-solax-modbus/pull/1286

empx92, 2025-03-24:
Have a X3-HAC-11 and could help, but at the moment it is hard to find time...until end of march

closed 2025-09-16 by github-actions[bot]