wills106/homeassistant-solax-modbus #1289: [Bug]: Sofar HYD 10KTL-3PH - Energy Storage Mode - changes are not being saved

lekom70, 2025-03-16:
### Describe the bug

Hello, I have the Solax integration version 2025.01.9  I have noticed an issue with newer versions for the Sofar HYD 10KTL-3PH inverter. Specifically, there is no response when trying to save changes to the Energy Storage Mode option.
There is no response when changing, for example, from Passive Mode to Self Use and vice versa. In version 2025.01.9, it works fine. However, after updating to later versions, it stops working.

### Integration Version

2025.03.4

### Homeassistant core version

2025.3.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES110

### Connection Method

RS485-USB

azeair, 2025-03-18:
Same here... 

[image]

[image]

azeair, 2025-03-18:
[image]

hubertozog, 2025-03-21:
Have same issue. I also lost ability to read second battery paramaters.

cschlipf, 2025-04-29:
Hm... can't reproduce. Still works for me. Are you still having this issue?

cschlipf, 2025-04-29:
> [image]

Are you using the LSE-3?

hubertozog, 2025-04-30:
@cschlipf All good for me, the problem was on my side.

alias, 2025-05-02:
Have this Problem of switching between the modes too; in rare cases it works, but mostly there is an modbus error.

hyd 20KTL 3PH and LSE3 Ethernet Modbs adapter
solax 2025.04.4 HA 2025.4.4 OS 15.5  HA Green

cschlipf, 2025-05-03:
@alias I found a better and more reliable way that works very well for me. This involves switching the LSE3 adapter into a different working mode and even works with the LSW3 adapter.

Please see the updated documentation: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-installation/#configuration

Can you please test this?

alias, 2025-05-11:
Thanks @cschlipf siwtching between modes seems to work instant and reliable after going to transparency mode. 

But many readings (like battery input) are coming now with strange (huge) values; a phenomen that occured rarely bevor. And With HA2025.05.1 (Core), I also seem to get errors like below. Switching back to ha2025.4.4 with Data Collection Mode now.

```
Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:707
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 9. Mai 2025 um 23:08:10 (82 Vorkommnisse)
Zuletzt protokolliert: 11:19:35

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 707, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 868, in async_read_modbus_registers_all
    if not await group.readPreparation(self.data):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/sensor.py", line 152, in readPreparation
    return await battery_config.check_battery_on_start(hub, old_data, key_prefix, batt_nr, batt_pack_nr)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_sofar.py", line 4004, in check_battery_on_start
    readed = decoder.decode_16bit_uint()
  File "/config/custom_components/solax_modbus/payload.py", line 370, in decode_16bit_uint
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:707
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 00:03:45 (2 Vorkommnisse)
Zuletzt protokolliert: 07:41:02

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 707, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 891, in async_read_modbus_registers_all
    if not await group.readFollowUp(self.data, data):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/sensor.py", line 164, in readFollowUp
    return await battery_config.check_battery_on_end(hub, old_data, new_data, key_prefix, batt_nr, batt_pack_nr)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_sofar.py", line 4027, in check_battery_on_end
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
```

alias, 2025-08-03:
Coming back to this after a while. Thx @cschlipf.

Crazy spikes continue to happen. This morning. I see around one time every few day some megawatts , tis morning "Export Energy Today" and then battery_input_energy_today (same time) reportet 40MWatts.

To the max-limit (I pit "20" (assuming kW) seems not to work for me.

As per your last comment: I have only one service on modbus.

BTW: Modbus Transparency increases this problem drastically, ist not usable for me. I had to switch back.

hyd 20KTL 3PH and LSE3 Ethernet Modbs adapter
solax 2025.07.7  HA Green

closed 2025-09-17 by github-actions[bot]