wills106/homeassistant-solax-modbus #1291: Growatt Modbus adress list

flopp999, 2025-03-17:
I have summaries some addresses for Growatt.
I have tested this against MID XH.
Maybe some addresses can be wrong.
If you find any error or you know any missing address, please reply.

https://1drv.ms/x/s!AltbSnsa1GwWhYIE7G-g9HWnXdg4vA?e=N6G1YA

flopp999, 2025-04-08:
no, I have no documentation, I have only tested and then confirm my thoughts :)

TonyMathiesen, 2025-04-08:
Nice work then 👍🏼

Stefan-Linux, 2025-04-13:
Hi,
I'm currently looking for information on Growatt inverters, which led me here.
Still, I cannot open the Excel Sheet, as I'm told that it is too large and I'm browsing the Web on Ubuntu... (does not change to "local" Excel automatically).
(I will try if I can download it somehow)

Maybe you can take this feedback and provide some other / additional access to the sheet for the future?

Thanks! :)
 Stefan

[image]

flopp999, 2025-04-13:
@Stefan-Linux thanks for sharing. It was my mistake. I had added a lot of data to the file, which I should not have done. It is now removed and the file size is now 0,5MB, before it was 150MB :)

flopp999, 2025-04-13:
@Stefan-Linux I have updated the code for Growatt in the integration, you can find the latest updated here, I will soon make a PR for this. But I need some feedback first
https://github.com/flopp999/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_growatt.py

Stefan-Linux, 2025-04-14:
Hi flopp999,
I can now open the Excel sheet, thanks!
But it may not be what I was looking for. I'm trying to get a local connection to a NEO-800M. It's already in the cloud, but I cannot access it with my homeassistant.

Once I have the access, the Excel might become useful.

Thanks, Stefan

MuadDib007, 2025-06-05:
I see the remark for some items:

`0-2500 charge, 65535-63035 means discharge. Divide by 10`
The proper way to address this field is as a signed integer (int16) which automatically accounts for the possibility of a negative value.
Some programming languages require you to define an int as signed if you need it to be.

Would the int be unsigned, then it can only hold 0 or a positive value.

Good work on making this list!