wills106/homeassistant-solax-modbus #1295: Initializing Solax integration in Home Assistant fails (S/N cannot be read)

weisen-p, 2025-03-19:
When setting up the Solax integration in Home Assistant, the initialization is not completed after I have made the configuration entries. I can see in the Home Assistant logs that the serial number (of the dongle, I assume) could not be found or read (config_entries.py line 753).

I have an Azzurro 3PH HYD 10000 ZSS inverter (identical to Sofar Solar HYD 10KTL-3PH) as master with an Azzurro 1PH 4600TLM-V3 as slave and Pylontech batteries (9.6 kWh) connected to the master. Communication with the inverter takes place via the WLAN stick logger (LSW-3) and NOT via the LSE-3 logger stick.

I entered the following in the Solax configuration: Interface: TCP / Ethernet // Modbus address of the inverter: 1 // Inverter type: sofar // Polling and timeout intervals: 15 and 5 s as preset // Emergency power switch (EPS), dry contact box and parallel mode NOT activated // IP address of the inverter: <IP address of the WiFi dongle> // TCP port: 8899 // tcp_type: Modbus TCP // Enable readout: yes

READING the registers with the WiFi stick logger works with the Solarman integrations (from both Stephan Laubert and David Rapan), but the serial number of the stick must already be entered during configuration. WRITING with the integration from David Rapan does NOT work.

Is it technically IMPOSSIBLE to WRITE to registers with the WLAN stick logger via Modbus?

[---]

Bei der Einrichtung der Solax Integration in Home Assistant wird die Initialisierung nicht abgeschlossen, nachdem ich die Eingaben zur Konfiguration getätigt habe. In den Homeassistant Logs sehe ich, dass die Seriennummer (des Dongles, nehme ich an) nicht gefunden bzw. ausgelesen werden konnte (config_entries.py Zeile 753).

Ich habe einen Azzurro 3PH HYD 10000 ZSS Wechselrichter (baugleich mit Sofar Solar HYD 10KTL-3PH) als Master mit einem Azzurro 1PH 4600TLM-V3 als Slave und am Master angeschlossenen Pylontech Batterien (9,6 kWh). Die Kommunikation mit dem Wechselrichter erfolgt über den WLAN Stick Logger (LSW-3) und NICHT über den LSE-3 Loggerstick.

Folgendes habe ich bei der Solax-Konfiguration eingegeben: Schnittstelle: TCP / Ethernet // Modbus-Adresse des Wechselrichters: 1 // Wechselrichter Typ: sofar // Polling und Timeout Intervalle: 15 bzw. 5 s wie voreingestellt // Notstromschalter (EPS), Dry Contact Box und Parallel Mode jeweils NICHT aktiviert // IP-Adresse des Wechselrichters: <IP-Adresse des WiFi Dongles> // TCP Port: 8899 // tcp_type: Modbus TCP // Auslesen aktivieren: ja

Das LESEN der Register mit dem WLAN Stick Logger funktioniert mit den Solarman Integrationen (sowohl von Stephan Laubert als auch von David Rapan), allerdings muss die Seriennummer des Sticks dort bereits bei der Konfiguration eingegeben werden. Das SCHREIBEN mit der Integration von David Rapan funktioniert NICHT.

Ist es technisch UNMÖGLICH, mit dem WLAN Sticklogger über Modbus in Register zu SCHREIBEN?

cschlipf, 2025-03-19:
Can you please post the serial number of your inverter (you can leave out the last 5 digits)

And no: Writing modbus registers is not possible with the WIFI stick logger as it cannot talk modbus TCP. See https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/#i-am-using-the-lsw-3-wifi-stick-logger-but-modbus-tcp-does-not-work

weisen-p, 2025-03-19:
The serial number of my inverter is ZP2ES010Pxxxxx.

cschlipf, 2025-03-19:
That serial number is handled by the integration:

https://github.com/wills106/homeassistant-solax-modbus/blob/bcd0488fe46eb53c211666eedf982b1e10c80056/custom_components/solax_modbus/plugin_sofar.py#L4072

So obviously the problem is that you are using the WIFI logger, which cannot establish a connection and therefore the serial number is not read at all. Please establish a supported connection as described above.

@wills106 - this issue can be closed.

cschlipf, 2025-05-03:
@weisen-p I have found a way that enables to use the LSW Wifi Logger to use with this integration. Please read the updated documentation: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-installation/#lsw-3-wifi-stick-logger