wills106/homeassistant-solax-modbus #1300: [Bug]: Inappropriate state_class: for Battery Input/Output Energy Today

romankysely, 2025-03-23:
### Describe the bug

I suggest changing state_class: total_increasing of the entities Battery Output Energy Today and Battery Input Energy Today to state_class: total. Both entities are reset to zero every day at midnight. This is also evident from the entity name ...Energy Today.
When using a single inverter in the Energy dashboard, this does not affect it, but when summing entities from multiple inverters, complications arise in the template. thanks

[image]

### Integration Version

2025.03.4

### Homeassistant core version

2025.3.3

### Inverter brand

Solax X3G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A12

### Inverter firmware versions

ARM 1.45, DSP 1.48

### Connection Method

Waveshare Adapter

### Dongle firmware

? LAN

Krawuzzi, 2025-05-13:
Hello @romankysely I'm having problems with the energy dashboard since the HA core update 2025.5.0
Output energy today is showing much smaller values than before.
Do you think the state_class is causing this problem?

[image]

I have opened a bug report here: https://github.com/home-assistant/core/issues/144798

romankysely, 2025-05-28:
Honestly, I don't know. I'm just getting ready for version 2025.5, so far I'm on 2025.3.4

romankysely, 2025-05-29:
Thx for advice.