wills106/homeassistant-solax-modbus #1302: Growatt, do you give permission

flopp999, 2025-03-24:
Hi, I am planning to create an integration for Growatt.
My idea is to fork your project and then remove everything except Growatt.

Do I get your permission to use your code?

wills106, 2025-03-24:
Is there a particular reason why you don't want to contribute to this project? It makes more sense to keep a shared codebase.

flopp999, 2025-03-24:
Sorry, I don't know what I was thinking, of course that is much better.
I will do some changes and test locally and then upload a PR.

Any ideas to change the name of the integration?

I don't what can be a common name.
Inverter Modbus
Solar Modbus, maybe not best because somebody may have only batteries.

The name is not very critical, but maybe people will find this integration easier with a name that gives an idea that it will work with more than Solax.

flopp999, 2025-03-24:
[image]
BMS 2 Module 1 is low because I added it today and it has not been charged.

rmk-sys, 2025-04-01:
That would be welcome to have more enteties for the bms (ARK XH) and batteries :-))

flopp999, 2025-04-01:
If you want to use the new entities, so far, can copy the text from the link and past it into the plugin_growatt.py in HA.
Please make a copy of the text that you have, before pasting the new text. If something goes wrong you can always go back to the copy you made.

https://raw.githubusercontent.com/wills106/homeassistant-solax-modbus/1ae033fc51c7efec3a8830f97ea1da2b30efc2f2/custom_components/solax_modbus/plugin_growatt.py

rmk-sys, 2025-04-02:
thx, I'd used your new entities (wow, a lot stuff). but for my system (ark xh) first generation (1 bms, 3 modules in a row, com version ZEBA) there are just 4 useable entities and I'd to deactivate more than 100. so I'd propose to create a bitmask suitable for the different systems out there. anyway, great work, i guess.

rmk-sys, 2025-04-02:
my MOD is GEN4 too. my thinking: possibly there are different com versions for the bms systems, which are usable just for the bitmask. now i think, i was wrong, because i remember, that nearly all the entities provided data, most of them '0' in my case.

if your plan to to provide more entities, i'm interested temp data and warn codes...

flopp999, 2025-04-02:
> my MOD is GEN4 too. my thinking: possibly there are different com versions for the bms systems, which are usable just for the bitmask. now i think, i was wrong, because i remember, that nearly all [...]

OK, yes if you do not have data for the entity it is better to disable it, but maybe first Enable and see if it works :)
Below is what I will be added when I am done, maybe more if I find any other :)

BMS 1 Module 1 Status
BMS 1 Module 1 SoC
BMS 1 Module 1 SoH
BMS 1 Module 1 Volt
BMS 1 Module 1 amp
BMS 1 Module 1 watt
BMS 1 Serial number
BMS 1 Monitoring Version text
BMS 1 Control version text

BMS 1 Module 1 Total Throughput
BMS 1 Module 1 Max Cell Volt
BMS 1 Module 1 Min Cell Volt
BMS 1 Module 1 Max Cell Temp
BMS 1 Module 1 Min Cell Temp
BMS 1 Module 1 Max Charge Current
BMS 1 Module 1 Max Discharge Current
BMS 1 Module 1 Real Capacity
BMS 1 Module 1 Warning Main Code
BMS 1 Module 1 Charge Cycle
BMS 1 Module 1 Serial number
BMS 1 Module 1 Control version text
BMS 1 Module 1 Control version number

Maybe you can help to get Monitoring/Control version number?
I don't know if I am checking correct register
Can you check what you get from your system if you do like below?
[image]
Can you also check those registers?
192
632
665
3101
3103
5899
5939

What versions do your system have? Can you send a picture like mine below?
This is my versions
[image]

rmk-sys, 2025-04-02:
growatt server does not provide any battery data for my system (anymore). but I got this from history data:

[image]

rmk-sys, 2025-04-02:
just these entities provide data:

[image]

[image]

this time i can't read out the registers, be cause of a growatt server error.

rmk-sys, 2025-04-02:
and this is logged in homeassistant:

[image]

flopp999, 2025-04-02:
> and this is logged in homeassistant:
> 
> [image]

Yes I noticed that. It is a mistake from me. It will be fixed today

Thanks for letting me know

rmk-sys, 2025-04-02:
Don't hurry, most of we're discussing here is 'nice to have'.  :-))

flopp999, 2025-04-02:
Already done and some new entities. Copt this copy to your HA
https://raw.githubusercontent.com/flopp999/homeassistant-solax-modbus/refs/heads/main/custom_components/solax_modbus/plugin_growatt.py

rmk-sys, 2025-04-02:
my register read out:

192  ->  3
632 ->  3
665 ->  3
3101 -> 5
3103 -> 5
5899 -> 0
5939 -> 0
5016-5018 -> 5-2078-5

thx, copied the new version. looks like HA can get data from entities bms 1 (but not module in name and every bms 2 entities)

rmk-sys, 2025-04-02:
I'm interestet in these entities (all delivered in history data):

BdcLlcTemp(℃)
bdcVersion (imho Firmware)
BmsMaxCellTemp(℃)
BmsMinCellTemp(℃)

but not in HA (example):

[image]

as i mentioned: there is no data for all entities with 'module' in naming.

and by the way for my system there are no  battery serial numbers provided (even not in history data).

thx for now.

flopp999, 2025-04-02:
@rmk-sys you have one BMS and 4 modules, right?
And ARK, correct?

Maybe ARK is not giving same data as APX, I have APX

Are you using the code I linked to or the code that was released this morning in 2025.04.1
BMS 1:1 max temp is working for me
[image]

> I'm interestet in these entities (all delivered in history data):
> 
> BdcLlcTemp(℃)
> bdcVersion (imho Firmware)
> BmsMaxCellTemp(℃)
> BmsMinCellTemp(℃)

do you mean that you see this datas on Growatt web or HA history data?

> and by the way for my system there are no battery serial numbers provided (even not in history data).

growatt web or HA?

Maybe your system does not gives data for battery modules, strange. Do you see any module data in ShinePhone?

rmk-sys, 2025-04-02:
i meant history data in growatt web server. sorry!

rmk-sys, 2025-04-02:
not even in shinephone.

rmk-sys, 2025-04-02:
ark xh is definitely different, battery modules working in a row. there is no sepearate bms for every module like yours.

flopp999, 2025-04-02:
@rmk-sys if you have Discord, can you add me, flopp999, then I can give you some tests and maybe we can solve the "problem" with ARK.

flopp999, 2025-04-03:
@rmk-sys
bdcVersion (imho Firmware):  can you try on web and check reg 3096 to 3103
3096-3097 is BDC(BMS) Monitoring software, in my case ZECA (23109,17217)
3098 is DTC, maye some kind of Diagnosic Code
3099,3100 is DSP Control version(BMS Control on web), in my case VDAA (22084, 16705)
3101 DSP version number, in my case 3
3103 BDC version number, in my case 3

What do you get?

rmk-sys, 2025-04-03:
bdcVersion (imho Firmware): can you try on web and check reg 3096 to 3103

[image]

3096-3097 is BDC(BMS) Monitoring software, in my case ZECA (23109,17217)
this is solved, i get it from 'bms_1_monitoring_version'

3098 is DTC, maye some kind of Diagnosic Code

[image]

3099,3100 is DSP Control version(BMS Control on web), in my case VDAA (22084, 16705)
this is what i'm looking for: has to be VCAA

[image]

rmk-sys, 2025-04-03:
3101 DSP version number, in my case 3

[image]

3103 BDC version number, in my case 3

[image]

rmk-sys, 2025-04-03:
these are the entities i get data already:

[image]
[image]

flopp999, 2025-04-03:
for DSP BMS Control version copy this code
https://raw.githubusercontent.com/flopp999/homeassistant-solax-modbus/refs/heads/main/custom_components/solax_modbus/plugin_growatt.py

rmk-sys, 2025-04-03:
thx - wrx. grt!

flopp999, 2025-04-03:
> Works great 👍 Thanks. Is there a reason why BMS 1 Module 1 Status is not there?

Do you mean that you have value 0 or you don't see the entity?
Have you enable the entity?
Have you replace the file with the code in this link?
https://raw.githubusercontent.com/flopp999/homeassistant-solax-modbus/refs/heads/main/custom_components/solax_modbus/plugin_growatt.py
What inverter do you have?
What battery do you have?
If you don't have GEN4 nothing of the new entities will be added, since it is not tested on any other GEN, than GEN4

rmk-sys, 2025-04-03:
final question for today: what is 'bms_1_module_1_tp' ? showing total discharge power for my system (also provided by total_battery_output_energy). right, or is it total charge + discharge power?

TonyMathiesen, 2025-04-03:
@flopp999 Maybe a bit unclear what I meant. The BMS 1 Module 1 Status is not included in the 2025.04.1 release, but the other BMS x Module x Status are.

flopp999, 2025-04-03:
> final question for today: what is 'bms_1_module_1_tp' ? showing total discharge power for my system (also provided by total_battery_output_energy). right, or is it total charge + discharge power?

No worries, just fun to add more entities :)

tp means Total Throughput BUT know I have more knowledge and that means what the battery/module has output(discharge)
I may change that name soon. I don't know why they

[image]

[image]

flopp999, 2025-04-03:
> [@flopp999](https://github.com/flopp999) Maybe a bit unclear what I meant. The BMS 1 Module 1 Status is not included in the 2025.04.1 release, but the other BMS x Module x Status are.

OK I see, correct. I have added more entities now after the release 04.1.
I will soon make a PR and then all new entities will be available for everybody.

If you want it now you can copy the code from the link and paste it to the file directly in HA.

flopp999, 2025-04-03:
Fixed the problem for not showing serial number

[image]

flopp999, 2025-04-04:
BdcLlcTemp(℃)
BmsMaxCellTemp(℃)
BmsMinCellTemp(℃)

On what addresses are those datas?

rmk-sys, 2025-04-04:
fw version :-))

BmsMaxCellTemp(℃) Solved!
BmsMinCellTemp(℃) Solved!

BdcLlcTemp(℃) -> BMS 1 Temp A: you got it!

and these are the data i can get from growatt server, if i  export history data set (excerpt of the bdc data) in a file:

[bms data.xls](https://github.com/user-attachments/files/19607360/bms.data.xls)

just this entity reports wrong:

[image]

again, great work and very busy, thx!

[image]

flopp999, 2025-04-04:
fw version ?? missing that one or what do you mean? :)

BMS 1 Status is what it reports after charge is complete. Even that the web doesn't say Standby modbus register days Standby. if you discharge it should change to status to Discharge

[image]

flopp999, 2025-04-04:
@rmk-sys 
fw version ?? missing that one or what do you mean? :)

BMS 1 Status is what it reports after charge is complete. Even that the web doesn't say Standby modbus register days Standby. if you discharge it should change to status to Discharge

[image]

rmk-sys, 2025-04-04:
ok,  I understand now and so everything is fine for me.

flopp999, 2025-04-04:
anyone that know on what address you can get Inverter Warn code or fault code?
I have tested PDF 1.24 addresses but it only gives me zeros(0).

rmk-sys, 2025-04-10:
IMHO: BDC -> Battery Device Controller. but anyway, in growatt terms it means the same.

flopp999, 2025-04-12:
Now I feal that I am ready with all changed and new entiteis. Please try latest code in below link. Can someone test it and confirm that it works without any errors? When that is done I will make a PR. 
Thanks
https://raw.githubusercontent.com/flopp999/homeassistant-solax-modbus/refs/heads/main/custom_components/solax_modbus/plugin_growatt.py

flopp999, 2025-04-13:
I need help to solve how to read the exact model from modbus. Can as many as possible read address 43 via the web?
I get 5400 I have MID 15KTL3 XH

rmk-sys, 2025-04-14:
my model is MOD7000TL3-XH and I get: 

[image]

flopp999, 2025-04-14:
> my model is MOD7000TL3-XH and I get:
> 
> [image]

thanks, OK same as me but with different model, so then that is wrong address.
Thanks

flopp999, 2025-04-14:
> my model is MOD7000TL3-XH and I get:
> 
> [image]

can you check address 3098?
I get 10001

rmk-sys, 2025-04-14:
3098 -> 0

flopp999, 2025-04-14:
OK, also wrong address. I will see if other address can help me. Thanks

flopp999, 2025-04-14:
can you check this

[image]

flopp999, 2025-04-14:
@rmk-sys can you check this?

[image]

rmk-sys, 2025-04-14:
[image]

[image]

flopp999, 2025-04-14:
@rmk-sys this

[image]

flopp999, 2025-04-14:
> [image]

interesting, 8262 means 15 and 8247 means 7
but the rest is same BUT i have MID and you have MOD. but MOD is between 3 and 10 kW, MID is 11 and 30, same functions but just a different range and capacity. So maybe Growatt doesnt care about the rest and just check address 127 to set correct model like MOD/MID etc

rmk-sys, 2025-04-14:
the max output power is stored in reg 121. 70 means 7k. 150 means 15k

flopp999, 2025-04-14:
> the max output power is stored in reg 121. 70 means 7k. 150 means 15k

can confirm this, i got data from 2 others. When check in PDF it says Mxxx Power, great :)

flopp999, 2025-04-14:
@rmk-sys I opened a new issue just for decoding of model #1324

flopp999, 2025-04-14:
this is what I am trying to do

[image]

flopp999, 2025-04-21:
I am ready to create a PR for my changes but I want some to test it before I do that.
Can you test this code, if you test it and it works please reply in this issue
Thanks

https://raw.githubusercontent.com/flopp999/homeassistant-solax-modbus/refs/heads/main/custom_components/solax_modbus/plugin_growatt.py

flopp999, 2025-04-23:
> it works for me, as far as i can see (can't test fault text), but i had to reintegrate some entities and edit my automations because of the entity renaming in your code.
> 
> all in all thx for the great work!

Thanks for testing.
Yes, I have renamed some entities and deleted some other. Some old entities was just getting data for BMS 1, they are renamed from old name to BMS 1.

flopp999, 2025-07-05:
I will add some more entries soon, Peak shaving information and some limits when charging from grid.
But first I will fix the other entries that I added a few months ago, but they are not implemented in the latest version of SolaX.
I will add a few entites per PR so it is not that difficult so see what I added.
@wills106 😄

closed 2025-09-16 by github-actions[bot]