wills106/homeassistant-solax-modbus #1307: [Bug]: 'Inverter Brand' - 'Inverter Model' Description of Error

cracyfloyd, 2025-03-28:
### Describe the bug

[image]

It seems to be that the switch (select) "Time 1 Active" not set the parameter in the inverter. See picture. 

### Integration Version

2025.03.4 

### Homeassistant core version

2025.3.4

### Inverter brand

Growatt Mod XH 6k

### Plugin used

plugin_growatt.py

### Serial prefix

DN1.0ZBDB

### Connection Method

Waveshare

rmk-sys, 2025-04-01:
I hope that will help. It works fine with my system MOD7000TL3-XH.

1. Clear time slots 
2. Take Your choice: select your time slots (note: no overlapping times accepted and nothing will be changed!)
3. Update Time Slots

TonyMathiesen, 2025-04-03:
Press the "Update Time Slots" to write the parameters to the inverter. Then it should be displayed on next read cycle. If overlapping time slots with the other time slots, then it can be rejected by the inverter.

HellixFix, 2025-04-05:
Had problem with updating slots when not all times correctly selected/initialized (slot 1,2,3). Selecting times for slot 1 only is not sufficient. After that all is OK.

cracyfloyd, 2025-04-05:
Thanks for all help. But nothing solved my problem.

the only what i get is an error when i push "update time slots"

See my picture the grey field:

[image]

TonyMathiesen, 2025-04-05:
Could you try to enable the entities for timeslots 2 & 3. Does that fix the error?

rmk-sys, 2025-04-06:
build a new dashbord like this:

[image]

and try it again.

cracyfloyd, 2025-04-17:
Anyone know when this bugfix will be released?

closed 2025-04-26 by wills106

wills106, 2025-04-26:
Just released [2025.04.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.2) with this change.

PR #1321 (2025-04-26): Fix error message if time slot x entities is not enabled.
For Growatt MOD inverters. Fix error message if time x entities is not enabled. 
Fix #1307