wills106/homeassistant-solax-modbus #1308: [FR]: Custom number of retries

robertugo2, 2025-03-30:
### Description

Hi,

I would like to suggest a new configuration parameter for specifying number of retires. For example in following line it is hard coded:
https://github.com/wills106/homeassistant-solax-modbus/blob/49c2a6d55a0c62e239e7dbdd6fd9a7d2662581d8/custom_components/solax_modbus/__init__.py#L262

It will be helpful for owners of LSW-3 adapter where it is known, that received response is incorrect and thus that multiply tries are performed when not needed. Also it is blocking pooling new data.

### Suggested Solution

Make number of retries configurable.

wills106, 2025-04-02:
I'll make this adjustable like I have done for timeout

robertugo2, 2025-04-02:
@wills106 thank you!