wills106/homeassistant-solax-modbus #1310: [Bug]: After upgrade 2025.3.4 yet another device was created in integration menu

PatrykW79, 2025-03-31:
### Describe the bug

I upgraded core to 2025.3.4, solax plugin was updated to 2025.03.4.
After that the device was duplicated or rather added the new device to the list.

[image]

The first one : X3-Hybrid-6.0
producent: SolaX Power
oprogramowanie: DSP v1.40 ARM v1.39 

The second one now: X3-Hybrid
producent: SolaX Power
oprogramowanie: DSP v1.40 ARM v1.39

That's the same inverter communication module, nothing has changed there just updates on HA.

The outcome is that new set of metrics were created so in energy dashboard I had to add new one with suffix _2 instead of sensor.solax_today_s_solar_energy (unavailable).

I must keep old and new metrics otherwise HA won't show historical data. Bummer.

### Integration Version

2025.03.4

### Homeassistant core version

2025.3.4

### Inverter brand

Corab Encore 6k

### Plugin used

plugin_solax.py

### Serial prefix

H34A06

### Inverter firmware versions

3.001.02

### Connection Method

Pocket_WIFI_V3

### Dongle firmware

09.03

### Detailed Error Log

No solax logs

JAFC-BandO, 2025-04-01:
Same issue. Really sucks

wills106, 2025-04-02:
Have you tried restarting HA again or removing both SolaX devices and adding in again?

I updated HA and didn't experience this issue and I'm now on Integration 2025.4.1 without any issues either.

Are there any errors in the log indicating to why it's split the configuration?

PatrykW79, 2025-04-02:
I did restart of services, I did the whole server, I restarted plugin.
About the logs I can check but that must be old log because in recent there is nothing from solax.
Which log should I search for?
I wanted to avoid any removal because I'm afraid that the rest of integration that relay on data will broke in addition not sure if the collected data will be erased or not.

Stoni06, 2025-04-02:
Hi, same issue. I completely uninstalled Solax Inverter Modbus also from HACS. Then I installed Solax Inverter Modbus 2025.4.1 and all works fine!
Best regards

PatrykW79, 2025-04-02:
@Stoni06 What with old data/historical data? Are they preserved?

Stoni06, 2025-04-02:
Yes, but I don't use historical data. Because during the faulty recording, I had incredible peaks of more than 15,000,000 kWh. And I've found out that all cumulative data in HA is far too inaccurate. Therefore, I only use the live display of power generation. Uninstallation and installation only took a few minutes.

Best regards

PatrykW79, 2025-04-05:
@Stoni06 I did the same and helped but it's just workaround not a solution.
Hope that will not happen once again but at least I know how to fix it. Thx!

closed 2025-04-05 by PatrykW79

Stoni06, 2025-04-07:
Thx PatrykW79. I think the problem is caused by an update to HA or the Solax Modbus add-in. I agree with you; we have a workaround for the future.