wills106/homeassistant-solax-modbus #1323: Different pooling?

flopp999, 2025-04-11:
I am doing some work for adding more entities for Growatt. Some of them, like SoH, Serial number, doesn't need to be pooled so often. Is there a way to set different pooling time for a few entities?
@wills106 @TonyMathiesen

TonyMathiesen, 2025-04-11:
Would make sense to differentiate on some of them as you mentioned. Don't know if this is already possible, there @wills106 might be able to advise 🙂

TonyMathiesen, 2025-04-29:
@flopp999 Maybe these can be used .. Have not tried it, but it seems that they could be set during the config flow of the integration. Then the user can reduce the time in the default scan, and we then have to add either medium or fast to the relevant sensors. 
https://github.com/wills106/homeassistant-solax-modbus/blob/a50cdc97962a6c04093ba4961705b6e3ace2d65e/custom_components/solax_modbus/const.py#L90-L92

closed 2025-09-16 by github-actions[bot]