wills106/homeassistant-solax-modbus #1327: [Bug]: Export control value is incorrect

Bl00d-B0b, 2025-04-18:
### Describe the bug

Export Control value is incorrect. looks like 10 times less

### Integration Version

2025.04.0

### Homeassistant core version

2025.4.2

### Inverter brand

X3-Hybrid-G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

DSP v1.50 ARM v1.47

### Connection Method

Lan3.0

### Dongle firmware

n/a

### Detailed Error Log

It's a problem with Export control Value.

When I set value in SolaX Cloud to 10.000, then I see in in solax-modbus plugin 1.000W.
When I set value in SolaX Cloud to 5.000, then I see in in solax-modbus plugin 0W.

when I set value in solax-modbus plugin 10.000, then i see it in a cloud 100.000....

Looks like value is 10 times higher!

Bl00d-B0b, 2025-04-30:
Hello,
Bug still exist in latest release. value in SolaX modbus plugin is 10 times less, then in device!

Ed

Bl00d-B0b, 2025-04-30:
[image]

[image]

Donach, 2025-06-21:
I can confirm the issue also when using:
X3-Ultra-30kW
by SolaX Power
Firmware: DSP 2 ARM 23.1
Hardware: Gen5 

Integration version 2025.06.1

wills106, 2025-06-21:
Probably needs an entry here:

https://github.com/wills106/homeassistant-solax-modbus/blob/5eed9c0a8b18ac62064532d5607c20b92b075c54/custom_components/solax_modbus/plugin_solax.py#L544-L558

Donach, 2025-06-23:
> Probably needs an entry here:
> 
> [homeassistant-solax-modbus/custom_components/solax_modbus/plugin_solax.py](https://github.com/wills106/homeassistant-solax-modbus/blob/5eed9c0a8b18ac62064532d5607c20b92b075c54/custom_components/solax_modbus/plugin_solax.py#L544-L558)
> 
> Lines 544 to 558 in [5eed9c0](/wills106/homeassistant-solax-modbus/commit/5eed9c0a8b18ac62064532d5607c20b92b075c54)
>  EXPORT_LIMIT_SCALE_EXCEPTIONS = [ 
>      ("H4", 10),  # assuming all Gen4s 
>      ("H34", 10),  # assuming all Gen4s 
>      ("H3UE", 10),  # Issue #339, 922 
>      ("H4372A", 1),  # Issue #857 
>      ("H4502A", 1),  # Issue #857 
>      ("H4502T", 1),  # Issue #418 
>      ("H4602A", 1),  # Issue #882 
>      ("H3BD", 10),  # X3-Ultra 
>      ("H3BF", 10),  # X3-Ultra 
>      ("H4752A", 1), 
>      ("H3BC", 10), 
>      ("H34B10H", 10),  # need return @jansidlo , 
>      #    ('H1E', 10 ), # more specific entry comes last and wins 
>  ]

The inverter is H3BC type, both of them are (master/slave).
What's weirder is that the master one has the 10x multiplication error on the export control - in the app, when I set 25000W, I get 2500W in HA, and if I set it to 25000W in HA, I get 250,000W! in the Solax app...
However the slave (same inverter type) does not have this weird issue.
Any clues? :-/

Bl00d-B0b, 2025-06-27:
just installed 2025.06.2 of plugin, still same issue. is it fixed? my device S/N is H34A.....

Donach, 2025-06-28:
> you can find all entities including disabled ones under settings > devices & services > entities tab In this list, you will find the config_export_control_limit_readscale. Click it and select enable this entity, confirm. After 30 seconds the entity will become visible.

Can confirm this has solved the issue for me.