wills106/homeassistant-solax-modbus #1332: Growatt decode energy

flopp999, 2025-04-20:
I need help to find how much energy/capacity that is connected to the inverter.

I have MID 15KTL3-XH, APX. On those addresses I have confirmed for my system, 5036 BMS 1 and 5076 BMS 2. This gives me 300 on both and I have checked those address before I connected more modules. So value divided by 10 gives the energy/capacity

flopp999, 2025-04-20:
@rmk-sys do you have a connection via Modbus TCP to your inverter and some knowledge of python?
If I give you a python code to record modbus registers, would you have the possibility to run that?
I am using Debian 12 and python3

If you or someone else could run that code for 2-3 minutes and then send me the data, I could maybe see on what register the kWh is written. It must be somewhere in the code

rmk-sys, 2025-04-20:
> flopp999

sorry, no coding experiences and my battery system is much more simple, there’s no information to about the capacity in history data on growatt web server. so, i think loading and unloading is just based on battery voltage and current.

closed 2025-07-25 by flopp999