wills106/homeassistant-solax-modbus #1335: [Bug]: 'solis ' - ' S5-EH1P3.6K-L' unable to initialize the integration post update eroors reading serial number

wymes, 2025-04-26:
### Describe the bug

Hi

After updating the integration none of the entities were being updated . the logs showed the following 
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-26 16:20:56.115 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis: reading serial number from address 0x80ec failed; other address may succeed
2025-04-26 16:20:56.115 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis: cannot find serial number, even not for other Inverter
2025-04-26 16:20:56.115 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis inverter type - serial number : unknown
2025-04-26 16:21:12.124 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-04-26 16:21:12.126 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 126 but got 0, IGNORING.
2025-04-26 16:21:12.127 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 62, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-26 16:21:12.128 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis: reading serial number from address 0x80ec failed; other address may succeed
2025-04-26 16:21:12.128 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read solis 0x80ec serial number: None, swapped: False
2025-04-26 16:21:12.128 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis: cannot find serial number, even not for other Inverter
2025-04-26 16:21:12.128 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis inverter type - serial number : unknown
2025-04-26 16:21:12.128 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec

Upon restoring a previous HA daily backup  all was well with the system again

 

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.4

### Inverter brand

solis

### Plugin used

plugin_solis.py

### Serial prefix

603102

### Connection Method

usr-w610 module 

### Additional context

I tried adding a new hub and that failed to initialize as well . Reverting to the previous HA backup  restored the integration to a working state updating entities

wills106, 2025-04-27:
When on 2025.04.2 have you tried deleting the Inverter and adding it back in again? I want to rule out people aren't on a very old config. As HA can cache old config flows. If you name the new setup exactly the same as the old, you won't loose any of your stats.
But it's always best to do a backup.
Do you have the timeout at the default of 5?

kelddamsbo, 2025-04-27:
On Growatt MOD6000 TL3-XH I have tried to delete both integration and the item in HACS, rebooted HA, added HACS element and integration again, but same problem again with version 2025.04.2
timeout at the default of 5
Gone back to 2025.04.1 and it works again

wymes, 2025-04-27:
> When on 2025.04.2 have you tried deleting the Inverter and adding it back in again? I want to rule out people aren't on a very old config. As HA can cache old config flows. If you name the new setup [...]

Hi  I opted to add a new hub  using different names  and the time out in the new and existing config was set to 5 . With regards to the new hub I created  it would just await initialization. deleted the new hub and created again a couple of times  with the same result

wymes, 2025-04-27:
Did you find the issue ??

mwrben, 2025-04-27:
Has this been inadvertently closed, there's no obvious understanding or fix of the issue?

chomtech, 2025-04-27:
> When on 2025.04.2 have you tried deleting the Inverter and adding it back in again? I want to rule out people aren't on a very old config. As HA can cache old config flows. If you name the new setup [...]

Yes, I created a snapshot in the VM beforehand so I could run multiple tests. Each time, I rolled the machine back, removed the integration, added it again with a different name—and every time, the same issue occurred: no communication with the inverter. Interestingly, when I downgraded (not the recommended way—via HACS) and restarted, everything worked without any issues.

jimbaxter, 2025-04-27:
I have been getting the same issues with my Solax G4 Inverter since the 2025.04.2 update.

It stopped connecting, I have tried switching from the modbus USB adaptor to using the PocketWifi 3.0 and it had the same issues, though in the logs below it seems Homeassistant is still using the USB modbus at /dev/ttyUSB1

```
2025-04-27 19:47:37.539 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2025-04-27 19:47:37.540 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x300 serial number: None
2025-04-27 19:47:37.544 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-04-27 19:47:37.547 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 11] Could not exclusively lock port /dev/ttyUSB1: [Errno 11] Resource temporarily unavailable
2025-04-27 19:47:37.648 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at /dev/ttyUSB1:0
2025-04-27 19:47:37.649 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 6672 -> Modbus Error: [Connection] Not connected[AsyncModbusSerialClient /dev/ttyUSB1:0]
2025-04-27 19:47:37.649 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 71, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-27 19:47:37.650 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2025-04-27 19:47:37.651 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x1a10 serial number: None
2025-04-27 19:47:37.651 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2025-04-27 19:47:37.651 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2025-04-27 19:47:37.651 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec
2025-04-27 19:47:47.653 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-04-27 19:47:47.656 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 11] Could not exclusively lock port /dev/ttyUSB1: [Errno 11] Resource temporarily unavailable
2025-04-27 19:47:47.758 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at /dev/ttyUSB1:0
2025-04-27 19:47:47.758 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2025-04-27 19:47:47.758 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-04-27 19:47:47.762 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 11] Could not exclusively lock port /dev/ttyUSB1: [Errno 11] Resource temporarily unavailable
2025-04-27 19:47:47.864 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at /dev/ttyUSB1:0
2025-04-27 19:47:47.864 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 0 -> Modbus Error: [Connection] Not connected[AsyncModbusSerialClient /dev/ttyUSB1:0]
2025-04-27 19:47:47.864 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 71, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-27 19:47:47.866 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2025-04-27 19:47:47.866 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number: None
2025-04-27 19:47:47.866 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-04-27 19:47:47.876 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 11] Could not exclusively lock port /dev/ttyUSB1: [Errno 11] Resource temporarily unavailable
```

raba-the-mark, 2025-04-27:
Hi, I have the same issue with Growatt SPH6000. Into the log i read this:

"reading firmware number from address 0x9 failed; other address may succeed"

I have been using USRCN USR-W610 connected with WiFi

Homeassistant core version
2025.4.4

PrimusNZ, 2025-04-27:
I am also getting this same issue, I have narrowed it down to  the following commit: [Ignore faulty Modbus reading](https://github.com/wills106/homeassistant-solax-modbus/commit/84be6b050126a6bf9c1995252d258df92f655578)

If I fully update to 2025.04.2 but use the `__init__.py` file from 2025.04.01 the addon operates as expected.

filipi992062, 2025-04-28:
> If I fully update to 2025.04.2 but use the `__init__.py` file from 2025.04.01 the addon operates as expected.

- I confirm that this solution works.

michaelbeelman, 2025-04-28:
> If I fully update to 2025.04.2 but use the `__init__.py` file from 2025.04.01 the addon operates as expected.

I also confirm this solution works for me. I'm running HA Core 2025.4.4 and SolaX integration 2025.04.2 on a SolaX Gen 1.

wymes, 2025-04-30:
I attempted the update to 2025.04.03 and had issues 

From the integration logs  

asyncio.exceptions.InvalidStateError: invalid state
2025-04-29 23:53:56.633 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 2 but got 0, IGNORING.
2025-04-29 23:54:02.026 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-04-29 23:54:02.030 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 1 but got 0, IGNORING.
2025-04-29 23:55:32.030 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 2 but got 0, IGNORING.

rolled back to the pre install backup restore service ok

wills106, 2025-05-01:
I have released [2025.04.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.4)

If you are still having issues ensure your RS485 - Ethernet adaptor is correctly setup if using one, the vast majority of them need to be set as Modbus over TCP and not Modbus RTU over TCP. Even though your Inverter is using RTU, you are communicating to the adaptor via Modbus over TCP and the adaptor is doing the RTU conversion.

wymes, 2025-05-01:
Installed  2025.04.4 and everything works fine again . 

Many thanks

adbastiaansen, 2025-05-01:
2025.04.4 working fine now! thank you wills106 !

wills106, 2025-05-01:
@GHMattZ can you provide logs like I have already requested. As we can't do anything with

> still not working

https://github.com/wills106/homeassistant-solax-modbus/issues/1343#issuecomment-2837639975

GHMattZ, 2025-05-01:
I updated again to the 04.4 version to make you the log files and now it works, I don't know were the problem was before, thank you for your help!

PrimusNZ, 2025-05-01:
Also confirming 2025.04.04 works as expected.

marmer1, 2025-05-02:
I still get the following for the evc (gen2 11kw). The inverter doesn't cause any issues:

```
Logger: homeassistant
Quelle: /usr/src/homeassistant/homeassistant/runner.py:112
Erstmals aufgetreten: 15:05:55 (36 Vorkommnisse)
Zuletzt protokolliert: 15:06:32

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 178, in callback_data
    used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 76, in processIncomingFrame
    data_len, pdu = self._processIncomingFrame(data[used_len:])
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 97, in _processIncomingFrame
    if (result := self.decoder.decode(frame_data)) is None:
                  ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py", line 112, in decode
    pdu.decode(frame[1:])
    ~~~~~~~~~~^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/register_message.py", line 263, in decode
    self.address, self.count = struct.unpack(">HH", data)
                               ~~~~~~~~~~~~~^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes
```

And

```
Logger: homeassistant
Quelle: /usr/src/homeassistant/homeassistant/runner.py:112
Erstmals aufgetreten: 15:05:55 (35 Vorkommnisse)
Zuletzt protokolliert: 15:06:33

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
```

ceemjay, 2025-05-19:
I am having the same problem too. It's so long since I set my Solis up I have forgotten the settings and cant find a definitive set of instructions re the configuration. Is anyone could send/post that would be great.

Is there a reload option for this add-on - I cant see one anywhere?

Is there any debug software to monitor modbus to eliminate HA config (though maybe that would need the same config!) Data is still going to Solis Cloud so at least some part of the dongle is working.

Happy to post more logs but these are the last few lines:
2025-05-19 16:49:38.308 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-05-19 16:49:38.308 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.149:502
2025-05-19 16:49:38.315 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.149', 502)
2025-05-19 16:49:38.416 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.149:502
2025-05-19 16:49:38.416 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Solis2: trying to determine inverter type
2025-05-19 16:49:38.416 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-05-19 16:49:38.417 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.149:502
2025-05-19 16:49:38.420 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.149', 502)
2025-05-19 16:49:38.523 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.149:502
2025-05-19 16:49:38.523 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 33004 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.149:502]
2025-05-19 16:49:38.524 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis2: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 62, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-05-19 16:49:38.532 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis2: reading serial number from address 0x80ec failed; other address may succeed
2025-05-19 16:49:38.532 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read Solis2 0x80ec serial number: None, swapped: False
2025-05-19 16:49:38.532 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis2: cannot find serial number, even not for other Inverter
2025-05-19 16:49:38.533 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis2 inverter type - serial number : unknown
2025-05-19 16:49:38.533 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec
2025-05-19 16:49:48.537 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.149', 502)

Cypheros-de, 2025-05-20:
QCells/Solax X3-Hybrid Gen4 , Waveshare RS485 To ETH (B) V1.486 Modbus.
Had the same problems after updating from a older 2023 version of wills106 custom component to 2025.04.4 . Many Modbus errors (error reading serial number..., etc. ) in the log and system freezes, Reboots. Nearly unusable. 

As wills106 recommended, I remove the old Solax device, restarted HA and added a new device with the same name as the old device and that did the trick. Modbus problems are gone.

ceemjay, 2025-05-25:
Thanks I had already removed my solis device and re-added it several times
to no avail which is why I was wondering if I had the config correct or
not. I have today removed it, rebooted the Pi and reinstalled the add-on
and still no joy,

Does the Solis inverter not talk to SolisCloud over Modbus as that is
working fine?

However I have since downloaded a modbus monitor program but it fails to
detect any activity from the dongle but as it is still communicating with
SolisCloud I feel there is a big gap in my modbus knowledge!

It has been fine for nearly 2 years an only started causing problems
recently

Thanks

durkimat, 2025-06-05:
2025.04.4 working on my Solis. Later versions broken (for me). Just spent hours getting back to something working including complete removal, and reinstall of the plugin.

Seqway, 2025-07-03:
@ceemjay 
any update here ?

closed 2025-09-16 by github-actions[bot]