wills106/homeassistant-solax-modbus #1336: [Bug]: 'Solis' - 'RHI-3P' Communication Loss with Inverter after Update to 2025.04.2

chomtech, 2025-04-26:
### Describe the bug

After updating the Solax Modbus integration to version 2025.04.2, I experienced a complete communication loss with my inverter model RHI-3P.
The system was working perfectly before the update.

Rolling back to the previous version (2025.04.1) immediately restores full functionality and stable communication with the inverter.
This suggests that the issue was introduced specifically in the 2025.04.2 update.

Please let me know if I can provide any additional logs or information to help troubleshoot the problem.

Thank you for your work on this integration!

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.4

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

7A1239

### Connection Method

RS485

### Detailed Error Log

2025-04-26 18:39:21.354 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 1 but got 0, IGNORING.
2025-04-26 18:39:21.355 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] PV: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 62, in async_read_serialnr
    if not inverter_data.isError():
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-26 18:39:21.392 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] PV: reading serial number from address 0x80ec failed; other address may succeed
2025-04-26 18:39:21.392 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] PV: cannot find serial number, even not for other Inverter
2025-04-26 18:39:21.392 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized PV inverter type - serial number : unknown

and every 10 seconds:

2025-04-26 18:39:31.439 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 2 but got 0, IGNORING.
2025-04-26 18:39:31.439 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] PV: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 62, in async_read_serialnr
    if not inverter_data.isError():
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-26 18:39:31.440 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] PV: reading serial number from address 0x80ec failed; other address may succeed
2025-04-26 18:39:31.440 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] PV: cannot find serial number, even not for other Inverter
2025-04-26 18:39:31.440 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized PV inverter type - serial number : unknown

kelddamsbo, 2025-04-26:
I have the same problem on my Growatt MOD6000 TL3XH

DidMyBest, 2025-04-26:
same for Solax Inverter. Does not connect via Mod bus TCP anymore since update zu 2025.04.2

MrRickwi, 2025-04-26:
Same for Solax device

TenFiliman, 2025-04-26:
I have the same problem as well with my Wattsonic Inverter.
After updating to version 2025.04.2 of the integration, all my inverter sensors became unavailable.
I also have RS485 Connection Method with up to date version of HA.

After rolling back to the previous version, all sensors work again.

I am getting all kinds of errors in log of HA. I hope it helps. If not sorry for spamming.

Logger: custom_components.solax_modbus.plugin_solinteg
Source: custom_components/solax_modbus/plugin_solinteg.py:71
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 8:44:54 PM (1 occurrences)
Last logged: 8:44:54 PM

wattsonic: attempt to read serialnumber failed at 0x2710
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solinteg.py", line 71, in _read_serialnr
    if not data.isError():
           ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'

Next error

Logger: custom_components.solax_modbus.plugin_solinteg
Source: custom_components/solax_modbus/plugin_solinteg.py:82
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 8:44:54 PM (1 occurrences)
Last logged: 8:44:54 PM

wattsonic: reading serial number from address 0x2710 failed; other address may succeed

Next error

Logger: custom_components.solax_modbus.plugin_solinteg
Source: custom_components/solax_modbus/plugin_solinteg.py:1289
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 8:44:54 PM (1 occurrences)
Last logged: 8:44:54 PM

wattsonic: cannot find serial number, even not for other Inverter

Next error

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:562
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 8:44:54 PM (2 occurrences)
Last logged: 8:44:54 PM

Modbus: ERROR: expected transaction 1 but got 0, IGNORING.
Modbus: ERROR: expected transaction 2 but got 0, IGNORING.

Next error

Logger: custom_components.solax_modbus.plugin_solinteg
Source: custom_components/solax_modbus/plugin_solinteg.py:93
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 8:44:54 PM (1 occurrences)
Last logged: 8:44:54 PM

wattsonic: attempt to read model failed at 0x2718
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solinteg.py", line 93, in _read_model
    if not data.isError():
           ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'

Last error

Logger: homeassistant
Source: custom_components/solax_modbus/plugin_solinteg.py:1294
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 8:44:54 PM (1 occurrences)
Last logged: 8:44:54 PM

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 329, in async_init
    self._invertertype = await self.plugin.async_determineInverterType(self, self.config)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solinteg.py", line 1294, in async_determineInverterType
    bh, bl = model // 256, model % 256
             ~~~~~~^^~~~~
TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'

filipi992062, 2025-04-26:
> same for Solax Inverter. Does not connect via Mod bus TCP anymore since update zu 2025.04.2

I have same problem with Solax

wills106, 2025-04-27:
Duplicate of #1335

closed 2025-04-27 by wills106