wills106/homeassistant-solax-modbus #1337: [Bug]:

filipi992062, 2025-04-26:
### Describe the bug

Tato chyba pochází z vlastní integrace.

Logger: custom_components.solax_modbus
Zdroj: custom_components/solax_modbus/__init__.py:562
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 00:40:02 (12 výskyty)
Naposledy logováno: 00:40:33

Modbus: ERROR: expected transaction 8 but got 0, IGNORING.
Modbus: ERROR: expected transaction 9 but got 0, IGNORING.
Modbus: ERROR: expected transaction 10 but got 0, IGNORING.
Modbus: ERROR: expected transaction 11 but got 0, IGNORING.
Modbus: ERROR: expected transaction 12 but got 0, IGNORING.

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.4

### Inverter brand

SOLAX X3 Hybrid G4

### Plugin used

plugin_solax.py

### Serial prefix

 H34C10

### Inverter firmware versions

DSP v1.46 ARM v1.44

### Connection Method

MODBUS

### Dongle firmware

	WIFI3.0

### Detailed Error Log

Tato chyba pochází z vlastní integrace.

Logger: custom_components.solax_modbus
Zdroj: custom_components/solax_modbus/__init__.py:562
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 00:40:02 (138 výskyty)
Naposledy logováno: 00:47:36

Modbus: ERROR: expected transaction 134 but got 0, IGNORING.
Modbus: ERROR: expected transaction 135 but got 0, IGNORING.
Modbus: ERROR: expected transaction 136 but got 0, IGNORING.
Modbus: ERROR: expected transaction 137 but got 0, IGNORING.
Modbus: ERROR: expected transaction 138 but got 0, IGNORING.

Tato chyba pochází z vlastní integrace.

Logger: custom_components.solax_modbus.plugin_solax
Zdroj: custom_components/solax_modbus/plugin_solax.py:86
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 00:40:02 (138 výskyty)
Naposledy logováno: 00:47:36

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed
SolaX: reading serial number from address 0x1a10 failed; other address may succeed

Tato chyba pochází z vlastní integrace.

Logger: custom_components.solax_modbus.plugin_solax
Zdroj: custom_components/solax_modbus/plugin_solax.py:71
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 00:40:02 (138 výskyty)
Naposledy logováno: 00:47:36

SolaX: attempt to read serialnumber failed at 0x0 data: None
SolaX: attempt to read serialnumber failed at 0x300 data: None
SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 71, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'

michaelbeelman, 2025-04-27:
I have a similar issue since the last update. I have a SolaX Power X3-MIC Gen 1 with serial MC502TF4172046. Hoping a (general) fix can be applied soon. Many thanks for maintaining this so well guys, much appreciated.

[image]

wills106, 2025-04-27:
Duplicate of #1335

closed 2025-04-27 by wills106