wills106/homeassistant-solax-modbus #1342: [Bug]: Large number of exceptions raised by pymodbus

jefft4, 2025-04-28:
### Describe the bug

Exception Error doing job: Fatal error: protocol.data_received() call failed. (None) has been raised 659 times in 2 days. Looks like it originates from pymodbus failing comms in some way.

Unfortunately, I can't tell which device is causing this - I have an inverter and an EVC using the integration. Both solax and solax_evc plugins are in use.

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.2

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H4502A (inverter)

### Inverter firmware versions

DSP v1.46 ARM v1.43

### Connection Method

SolaX wi-fi dongle

### Dongle firmware

App shows 3.017.02, not sure where to get internal

### Detailed Error Log

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:112
First occurred: 27 April 2025 at 08:58:37 (659 occurrences)
Last logged: 17:27:14

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
```

marmer1, 2025-04-28:
I am experiencing the same issue, but in my case, it is specifically related to the EVC.
EVC: X3-HAC Gen2 11kw

wills106, 2025-04-29:
Released [2025.04.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.3), does this fix your issue?

marmer1, 2025-04-30:
I still get the following:

Logger: homeassistant
Quelle: /usr/src/homeassistant/homeassistant/runner.py:112
Erstmals aufgetreten: 06:04:49 (63 Vorkommnisse)
Zuletzt protokolliert: 06:05:52

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state

Logger: homeassistant
Quelle: /usr/src/homeassistant/homeassistant/runner.py:112
Erstmals aufgetreten: 06:04:48 (62 Vorkommnisse)
Zuletzt protokolliert: 06:05:52

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 178, in callback_data
    used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 76, in processIncomingFrame
    data_len, pdu = self._processIncomingFrame(data[used_len:])
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 97, in _processIncomingFrame
    if (result := self.decoder.decode(frame_data)) is None:
                  ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py", line 112, in decode
    pdu.decode(frame[1:])
    ~~~~~~~~~~^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/register_message.py", line 263, in decode
    self.address, self.count = struct.unpack(">HH", data)
                               ~~~~~~~~~~~~~^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes

jefft4, 2025-04-30:
> Released [2025.04.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.3), does this fix your issue?

Sadly, not.
```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:112
First occurred: 08:04:15 (205 occurrences)
Last logged: 18:06:18

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
``` ... 206 times since the update. Do we know what causes an invalid state on the modbus socket?

wills106, 2025-05-01:
Closing down as duplicate / related to #1335

closed 2025-05-01 by wills106