wills106/homeassistant-solax-modbus #1343: [Bug]: 'Growatt' - 'SPH 4000' no values in HA since update to 2025.04.2

GHMattZ, 2025-04-28:
### Describe the bug

After the Update, I don't get any values shown at HA. It says "Diese Entität wird nicht mehr von der „solax_modbus"-Integration bereitgestellt. Wenn die Entität nicht mehr verwendet wird, lösche sie in den Einstellungen." It worked nearly perfect before in the last 6 Month. Sorry I'm not an expert with logfiles and so on. I don't know how to provide them.
EVCC could read the modbus signals, so the problem is not the inverter or the rs485 to ethernet converter.

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.4

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

RAAA212006

### Connection Method

RS485 to Ethernet

Habi1989, 2025-04-29:
Same for my SPH4600 , no working entities since update.
Even a restored Backup does not bring back the function.

BR
Habi

GHMattZ, 2025-04-29:
At my system, a restore of an old backup worked. With 2025.04.1 it works.

wills106, 2025-04-29:
Is this the same issue as #1335 hard to tell as you haven't provided any [error logs](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log).

GHMattZ, 2025-04-29:
Yes I think it's the same. Like I said I'm not an HA expert, so I don't know how to do this Log files , sorry. Also I don't know how I could try this solution mentioned "If I fully update to 2025.04.2 but use the __init__.py file from 2025.04.01 the addon operates as expected."
I'm happy at the moment that it works with the old backup.
 For you as an additional inormation, my config is not too old, I set up my system 5 month ago. As I tried to config a new inverter with the newest version, it also failed, it get stuck at the "initialization".

Habi1989, 2025-04-29:
My Error Logs with the newest version:

[image]

adbastiaansen, 2025-04-29:
Same problem here on 2x Growatt MIN 3000TL-XH. Installed previous version 2025.04.1 and all working fine!

GhostGum, 2025-04-29:
I experienced the same issues after latest update. 
Reverting this commit: https://github.com/wills106/homeassistant-solax-modbus/commit/04456ff23b0dea6091a656d532ff6245350f41cc solved my issues. .
I didn't look into it further, sorry. 
I hope that helps.

wills106, 2025-04-29:
Released [2025.04.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.3) should fix the issue.

GHMattZ, 2025-04-29:
I made the update to 2025.04.3 5 minutes ago. Unfortunately it don't solve my problem. 

Here the logs:

Logger: custom_components.solax_modbus.plugin_growatt
Quelle: custom_components/solax_modbus/plugin_growatt.py:7718
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 23:36:58 (14 Vorkommnisse)
Zuletzt protokolliert: 23:39:11

SPH: cannot find firmware version, even not for other Inverter

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus.plugin_growatt
Quelle: custom_components/solax_modbus/plugin_growatt.py:7756
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 23:36:58 (14 Vorkommnisse)
Zuletzt protokolliert: 23:39:11

unrecognized SPH inverter type - firmware version : unknown

adbastiaansen, 2025-04-30:
Still same problem on 2x Growatt MIN 3000TL-XH. Installed back to previous version 2025.04.1 and all is working fine again.

davidstrobl, 2025-04-30:
The same on Solax X3 G4, 2025.04.3 did not solve the problem.

Laksepote, 2025-05-01:
Same here with Growatt SPH3600. Reinstalling backup and avoiding future updates for a while

wills106, 2025-05-01:
Closing down as duplicate / related to #1335

closed 2025-05-01 by wills106