wills106/homeassistant-solax-modbus #1344: [Bug]: 'Azzurro/Sofar HYD ZSS' Strange time shift in the recordings

Hetzi74, 2025-04-29:
### Describe the bug

I have a strange issue at a Azzurro/Sofar inverter!

The values are correct in themselves. But the recording is as if there is an ever-increasing time difference.

[Image](https://github.com/user-attachments/assets/27c2622e-126d-429d-8310-2b7d4c4ba698)

The picture shows production on a sunny day. I restarted the Home Assistant with the red arrows. After the restart, the values are correct so far, but the recording and reality drift apart in time. After restarting again in the afternoon, the same applies. Sunset for me is currently at 20:00, but production still appears in the recording after 21:00!

What could be the reason for this behavior?

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.4

### Inverter brand

Azzurro/Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

ZP1ES020N...

### Connection Method

LSE-3 LAN Stick

cschlipf, 2025-04-29:
What does the inverter system time say? Is this in sync with your real time?

Note that the LSE-3 stick does to sync your inverters system time with a time server.

I use an automation that does that regularly instead:

```yaml
alias: Update Sofar System Time
description: ""
triggers:
  - trigger: time
    at: "00:00:00"
conditions: []
actions:
  - action: button.press
    metadata: {}
    data: {}
    target:
      entity_id: button.sofar_sync_rtc
mode: single

```

Hetzi74, 2025-04-29:
> server

As far as i know the system time of the inverter is correct. The monitoring at the Azzurro Cloud is correct and it deffinitly doesnt do this time of time shifting. Like 1h realtime is lasting at the home assisting minitoring around 20% longer or something of that scale.

ooJaan, 2025-04-30:
I am using the HYD10KTL-3P and having the same problem. I have noticed that the system time on my unit seems to drift pretty badly (~2min / hr). I am also using the Waveshare RS232/484 to eth converter. 
Whenever I reload the hub in homeassistant, the values seem to be up to date again, regardless of system time! For testing I even created an automation to sync the time every 15 minutes (dont worry it only ran for one hour), and even then the problem still remained. So it seems like the system time is not the problem here. The drifting of it may still be tho. 
What time is used for the sensor in homeassistant? The system time of the inverter, the time from homeassistant or the inverter time + offset to the homeassistant time which is only calculated at the start?

cschlipf, 2025-04-30:
I also have a Sofar HYD 10 KTL-3P. However I cannot reproduce this, so all I can do is some guess work what happens on your end. In the end system time shouldn't matter. HA should write the time at the time it fetches it through modbus. So in theory the HA system time should be what counts. I am not aware that time is read from ModBus (but could be wrong on this).

Really strange.

Just checked the system time (that entity is disabled by default, but you can enable it), which was synchronized last Sunday for the last time:

[image]

I don't see a measurable drift in my case.

Hetzi74, 2025-05-01:
> What does the inverter system time say? Is this in sync with your home assistant system time?

Ok, that could be the problem!

I have activated now the System Time Entity now, and that replicates that behavior!

After a Homeassistant reboot, i have HA-Time (the real time) and Azzurro System time nearly in sync (a few sec off) after 10min, the Azzurro is already 5 min behind!

[image]

OK ill have to take a look to that network stick and see if i can set something about the Timeserver!

cschlipf, 2025-05-01:
Woah... that's hefty.

ooJaan, 2025-05-02:
Ok so I enabled the system time and created a helper to show the timediff and let it run for a day. This seems to be the result:

[image]

It dropped to near zero when I reloaded the hub. I synced the time of the inverter at the same time I reloaded the configuration. And over that day the time on the display "only" drifted a little more than an hour, while the reported time via modbus in homeassistant drifted ~15 hours.
This is the template for the helper:
```
{% set inverter = strptime(states.sensor.sofar1_rtc.state, "%Y-%m-%d %H:%M:%S").replace(tzinfo=now().tzinfo) %}
{{ (now() - inverter).total_seconds() }}
```

cschlipf, 2025-05-02:
Wondering why. Is this a faulty system clock in your inverter?

ooJaan, 2025-05-02:
I am using the waveshare rs232/485 to eth adapter. I am not sure if the system time of the inverter is the problem here, since it "only" drifted 1h, but the reported time and all other values via the integration are behind 15h. The time on my HaOs seems perfectly in sync.

cschlipf, 2025-05-02:
@ooJaan in combination with a LSW3 WiFi Dongle?

ooJaan, 2025-05-02:
For that test I did not. I plugged it in now.

cschlipf, 2025-05-02:
Because the LSW should do an NTP time sync.

Hetzi74, 2025-05-02:
> Wondering why. Is this a faulty system clock in your inverter? Are both of you using an LSE-3?

Yes Iam using a Ethernet connection over a LSE-3 stick.

I wanted to look up how to set the time on that stick, but could not find anything time related on its website.

Could another version of the firmware help for it? I have currently ME_0E_270A_1.11 on it. But i dont know where to get a different firmware version for that stick.

On the Inverter itself i have a correct time, visible on the screen of the inverter. Whatever time this is what i get in that now enabled sensor.azzurro_rtc sensor, its not the time from the inverter display itself.

Would there be a way to tell homeassistant to ignore that strange LSE-3 sticker time on its logging and just use the correct local time to store the values from the inverter?

cschlipf, 2025-05-02:
1.11 is causing quite some problems. See the Sofar FAQ. You can get in touch with the SolarMan support and ask them to downgrade it remotely to 1.09. The are pretty responsive.

Hetzi74, 2025-06-04:
Maybe its solved now, but was a quite a way. The Solarman support responded to me that i should contact the seller of the installation for it. The Azzurro support was very friendly and overall helpful, but it was not possible to downgrade the LSE-3 stick. It was possible to apply the file, the stick told to restart and did come back online with the same 1.11 Version. Was the same for the Azzurro support doing it remotely or for me doing it in the local network with the provided file.

Finally Azzurro sent me a replacement LSE-3 stick with Version 1.0E. 1st I had to reset that thing to get access to it over the local network, it was a already installed one at some other environment. Since yesterday I got it working and it seems to provide the data a lot better!