wills106/homeassistant-solax-modbus #1348: [Bug]: 'Inverter Brand' - 'Inverter Model' Description of Error

rulez86, 2025-04-29:
### Describe the bug

Dear all,

i love your ha addin and it works fine all days till today. I tried to update to new version 2025.04.2 but unfortunatelly, for some reason i do not receive any data from my growatt mod tl6000-xh inverter. With the version before (2025.04.1), everything works fine.

Now, i rolled back to 2025.04.1 and it's all good.

Do i have to do any special setting after updating or is there a bug? Let me know if i can help to find out the problem. I love this addin and it would be great to have in running smoothly, also with new update.

cheers,
rulez86

### Integration Version

2025.04.2

### Homeassistant core version

2025.4.4

### Inverter brand

Growatt MOD6000TL-XH

### Plugin used

plugin_growatt.py

### Serial prefix

DNAA045100

### Connection Method

USB - ModBus

TonyMathiesen, 2025-04-29:
Hi @rulez86 Will be fixed with PR #1347. Use 2025.04.01 until new version is released 😉

wills106, 2025-04-29:
Released [2025.04.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.3) containing this fix.

rulez86, 2025-04-29:
Thanks a lot for quick fix. I have two inverters. One with battery and another one without. Will test it tomorrow as soon as my second inverter wakes up from standby at sunrise.

rulez86, 2025-04-30:
Good Morning, unfortunately it's still not working with new release 2025.04.3. Any idea? I have two inverters. A Growatt MOD 6000TL-XH and a MIC3300TL-X inverter.

Here some log information. May that helps fixing the issue:
```
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-30 08:50:18.217 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: reading firmware number from address 0x9 failed; other address may succeed
2025-04-30 08:50:18.217 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt 0x9 firmware number before potential swap: None
2025-04-30 08:50:18.217 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: trying alternative location
2025-04-30 08:50:18.234 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 1 but got 0, IGNORING.
2025-04-30 08:50:18.234 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: attempt to read firmware failed at 0x9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 65, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-30 08:50:18.234 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: reading firmware number from address 0x9 failed; other address may succeed
2025-04-30 08:50:18.234 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read MIC3300TL-X 0x9 firmware number before potential swap: None
2025-04-30 08:50:18.234 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: trying alternative location
2025-04-30 08:50:18.270 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 2 but got 0, IGNORING.
2025-04-30 08:50:18.270 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: attempt to read firmware failed at 0xbb9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 65, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-30 08:50:18.270 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: reading firmware number from address 0xbb9 failed; other address may succeed
2025-04-30 08:50:18.270 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt 0xbb9 firmware number before potential swap: None
2025-04-30 08:50:18.270 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: cannot find firmware version, even not for other Inverter
2025-04-30 08:50:18.270 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized Growatt inverter type - firmware version : unknown
2025-04-30 08:50:18.270 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec
2025-04-30 08:50:18.284 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 2 but got 0, IGNORING.
2025-04-30 08:50:18.284 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: attempt to read firmware failed at 0xbb9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 65, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-30 08:50:18.285 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: reading firmware number from address 0xbb9 failed; other address may succeed
2025-04-30 08:50:18.285 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read MIC3300TL-X 0xbb9 firmware number before potential swap: None
2025-04-30 08:50:18.285 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: cannot find firmware version, even not for other Inverter
2025-04-30 08:50:18.285 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized MIC3300TL-X inverter type - firmware version : unknown
2025-04-30 08:50:18.285 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec
2025-04-30 08:50:28.271 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: trying to determine inverter type
2025-04-30 08:50:28.286 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] MIC3300TL-X: trying to determine inverter type
2025-04-30 08:50:28.312 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 3 but got 0, IGNORING.
2025-04-30 08:50:28.312 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: attempt to read firmware failed at 0x9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 65, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'
2025-04-30 08:50:28.312 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: reading firmware number from address 0x9 failed; other address may succeed
[... cut ...]
2025-04-30 08:51:14.977 ERROR (MainThread) [frontend.js.modern.202504110] Uncaught error from Safari 18.3.1 on macOS 10.15.7
Error: Failed to execute 'define' on 'CustomElementRegistry': the name "weather-radar-card-editor" has already been used with this registry
message (node_modules/core-js/modules/web.dom-exception.stack.js:25:24)
src/scoped-custom-element-registry.ts:181:8
/local/community/weather-radar-card/weather-radar-card.js:29:67
n (/local/community/weather-radar-card/weather-radar-card.js:1:915)
module code (/local/community/weather-radar-card/weather-radar-card.js:1291:141)
```

rulez86, 2025-04-30:
@PatrikTrestik 
I read your comments in PR #1347. So, i think we still have to wait for your fix, right?

PatrikTrestik, 2025-04-30:
PR #1351

rulez86, 2025-04-30:
@PatrikTrestik thanks for this Info. So, we wait for merge and Release by @wills106 in new Version.

wills106, 2025-05-01:
Closing down as duplicate / related to #1335

closed 2025-05-01 by wills106

rulez86, 2025-05-01:
2025.04.4 works fine for me, thanks.