wills106/homeassistant-solax-modbus #1349: [Bug]: Modbus still not working

MrRickwi, 2025-04-30:
### Describe the bug

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:562
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 06:10:53 (42 Vorkommnisse)
Zuletzt protokolliert: 06:13:06

Modbus: ERROR: expected transaction 38 but got 0, IGNORING.
Modbus: ERROR: expected transaction 39 but got 0, IGNORING.
Modbus: ERROR: expected transaction 40 but got 0, IGNORING.
Modbus: ERROR: expected transaction 41 but got 0, IGNORING.
Modbus: ERROR: expected transaction 42 but got 0, IGNORING.

### Integration Version

2025.04.3

### Homeassistant core version

2025.4.4

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

000000

### Inverter firmware versions

?

### Connection Method

Modbus

### Dongle firmware

?

MrRickwi, 2025-04-30:
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:562
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 06:10:53 (126 Vorkommnisse)
Zuletzt protokolliert: 06:17:48

Modbus: ERROR: expected transaction 122 but got 0, IGNORING.
Modbus: ERROR: expected transaction 123 but got 0, IGNORING.
Modbus: ERROR: expected transaction 124 but got 0, IGNORING.
Modbus: ERROR: expected transaction 125 but got 0, IGNORING.
Modbus: ERROR: expected transaction 126 but got 0, IGNORING.

MrRickwi, 2025-04-30:
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:71
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 06:10:53 (126 Vorkommnisse)
Zuletzt protokolliert: 06:17:48

SolaX: attempt to read serialnumber failed at 0x0 data: None
SolaX: attempt to read serialnumber failed at 0x300 data: None
SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 71, in async_read_serialnr
    if not inverter_data.isError():
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'isError'

MrRickwi, 2025-04-30:
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:7740
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 06:10:55 (42 Vorkommnisse)
Zuletzt protokolliert: 06:17:48

SolaX: cannot find any serial number(s)

MrRickwi, 2025-04-30:
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:8038
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 06:10:55 (42 Vorkommnisse)
Zuletzt protokolliert: 06:17:48

unrecognized inverter type - serial number : unknown

MrRickwi, 2025-04-30:
I‘m using Modbus over USB

adbastiaansen, 2025-04-30:
Same problem on Growatt MIN 3000TL-XH. Installed back to previous version 2025.04.1. (Modbus over USB)

filipi992062, 2025-04-30:
If I fully update to 2025.04.3 but use the __init__.py file from 2025.04.01 the addon operates as expected.

I confirm that this solution works. (Solax X3 Hybrid G4)

wills106, 2025-05-01:
Closing down as duplicate / related to #1335

closed 2025-05-01 by wills106