wills106/homeassistant-solax-modbus #1353: [Bug]: Always need to restart HA twice to get a modbus connection

rudibarani, 2025-05-01:
### Describe the bug

About half a year ago I noticed an error when restarting HA after updating an integration. I would always get this error message after the restart:

> **Invalid config**
> The following integrations and platforms could not be set up:
> 
> [modbus](https://www.home-assistant.io/integrations/modbus) ([Show logs](http://server.rudibarani.de:8321/config/logs?filter=modbus))
> [solax_modbus](https://homeassistant-solax-modbus.readthedocs.io/) ([Show logs](http://server.rudibarani.de:8321/config/logs?filter=solax_modbus))
> Please check your config and [logs](http://server.rudibarani.de:8321/config/logs).

**A second, complete restart of HA solves this issue and everything works fine.** 
Is anybody else experiencing this problem?

### Integration Version

2025.04.4

### Homeassistant core version

2025.4.4

### Inverter brand

Q-Cells (Solax)

### Plugin used

plugin_solax.py

### Serial prefix

H34B15

### Inverter firmware versions

ARM: 1.47-1.09 DSP: 1.5

### Connection Method

Waveshare Adapter

### Dongle firmware

V1. 486

### Detailed Error Log

Modbus Log shows the following messages:

> **Home Assistant Core**
>
>  Entity sensor.conjunction_reactive_power (<class 'custom_components.chint_pm.sensor.ChintPMModbusSensor'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'chint_pm' custom integration 
>  14:05:00 – (WARNUNG) Sensor - Meldung ist erstmals am 14:04:29 aufgetreten und erscheint 5-mal
>  Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'register_message' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu.py) 
>  14:04:05 – (FEHLER) setup.py
>  Setup failed for 'modbus': Unable to import component: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py) 
>  14:03:05 – (FEHLER) setup.py
>  Detected blocking call to import_module with args ('homeassistant.components.modbus',) in /usr/src/homeassistant/homeassistant/loader.py, line 1052: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 712, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 683, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2040, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 171, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1004, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1052, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path) 
>  14:03:05 – (WARNUNG) util/loop.py

Solax Log shows this message:

> **Home Assistant Core**
> 
>  Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'register_message' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu.py) 
>  14:04:05 – (FEHLER) setup.py

HA-Supi, 2025-05-03:
Maybe your Waveshare Settings?

[image]
You can also Update your Waveshare device… current FW 1.523, See here:
https://www.photovoltaikforum.com/wissen/entry/86-datenaustausch-mit-home-assistant-%C3%BCber-rs-485-com/

rudibarani, 2025-05-13:
Thanks for your idea. It seems, that was not the cause. 

I just had a look into the developer tools when experiencing this problem. There was this message:

`Component error: modbus - cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)`

Maybe this helps you to pinpoint how this error occurs and could be solved.

closed 2025-09-16 by github-actions[bot]