wills106/homeassistant-solax-modbus #1360: [Bug]: Growatt - MIN 6000TL-XH - No connection to Modbus since 2025.03.04

uspino2, 2025-05-06:
### Describe the bug

No connection to Modbus since 2025.04.1. Tried restarting module, clearing cache etc., to no avail. Downgrading to 2025.03.4 works again. 

### Integration Version

2025.04.4

### Homeassistant core version

2025.4.4

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

AL1

### Connection Method

Modbus HF5111B

### Detailed Error Log

2025-05-06 17:57:24.572 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x13bb bms_2_monitoring_version
2025-05-06 17:58:15.019 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 17:58:15.019 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5011 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 17:58:50.199 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 17:58:50.199 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5400 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 17:59:25.421 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 17:59:25.421 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5520 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:00:00.916 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:00:00.916 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5640 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:00:35.928 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:00:35.928 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5011 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:01:10.939 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:01:10.940 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5011 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:01:46.443 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:01:46.443 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5760 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:02:20.075 ERROR (MainThread) [custom_components.visonic.pyvisonic] [_sequencer] Visonic Plugin has suspended all operations, there is a problem with the communication with the panel (i.e. data has not been received from the panel in 0:04:00.715371)
2025-05-06 18:02:21.633 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:02:21.633 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5400 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:02:56.800 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:02:56.800 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5400 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:03:31.813 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:03:31.813 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5011 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:03:53.760 ERROR (MainThread) [homeassistant.components.shelly] Error fetching Estor 1 data: An error occurred while retrieving data from Estor 1
2025-05-06 18:04:06.826 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:04:06.827 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5011 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:04:42.291 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:04:42.292 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5881 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:05:17.512 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:05:17.513 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5520 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:05:52.741 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:05:52.742 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5520 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:06:27.920 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:06:27.920 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5400 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
2025-05-06 18:07:03.094 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
2025-05-06 18:07:03.094 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 5400 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request

uspino2, 2025-05-07:
Also, I tried updating my Waveshare RS485 TO ETH (B) to 1.523 and I think I bricked the device. I can factory reset shortcutting NC, and then access the device through Vircom, but any parameter change -including Network settings- crashes the module and turns Link down.

coen1111, 2025-05-07:
I can confirm that also for my Growatt min3600tl-xh the latest version is not working. Downgrading to 2025.04.1 fixed it.

uspino2, 2025-05-19:
Got a new Waveshare module, this time I was able to upgrade to v1.523 (my old module got bricked in the process) and all is good again.

closed 2025-05-19 by uspino2