wills106/homeassistant-solax-modbus #1365: [Bug]: 'Growatt' - 'MIN 3000TL-XH' read failed at 0x13b2: bms_2_serialnumber

maartentamboer, 2025-05-10:
### Describe the bug

Component continuously spams the log with the message `pv: read failed at 0x13b2: bms_2_serialnumber`. Also during `__init__` multiple tracebacks can be seen which may have to do something with this. See Detailed Error Log for more information.

### Integration Version

2025.04.4

### Homeassistant core version

 2025.5.0

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

AL1.0ZA

### Connection Method

RS485 TO POE ETH (B)

### Detailed Error Log

```log
2025-05-10 11:50:43.915 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 736, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa6 in position 3: ordinal not in range(128)
2025-05-10 11:50:58.914 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 736, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa6 in position 3: ordinal not in range(128)
2025-05-10 11:51:13.915 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 736, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa6 in position 3: ordinal not in range(128)
2025-05-10 11:51:28.915 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 736, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xa6 in position 3: ordinal not in range(128)
2025-05-10 11:51:43.918 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
2025-05-10 11:51:58.920 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
2025-05-10 11:52:13.916 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
2025-05-10 11:52:28.921 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
2025-05-10 11:52:43.920 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
2025-05-10 11:52:58.921 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
2025-05-10 11:53:13.922 WARNING (MainThread) [custom_components.solax_modbus] pv: read failed at 0x13b2: bms_2_serialnumber 
```

maartentamboer, 2025-05-10:
For now I've commented out this part in `plugin_growatt.py`. The log messages are no longer happening.

```python
    # GrowattModbusSensorEntityDescription(
    #     name = "BMS 2 Serial Number",
    #     key = "bms_2_serialnumber",
    #     register = 5042,
    #     unit = REGISTER_STR,
    #     wordcount=8,
    #     entity_registry_enabled_default = False,
    #     allowedtypes = GEN4,
    #     entity_category = EntityCategory.DIAGNOSTIC,
    #     icon = "mdi:information",
    # ),
```

flopp999, 2025-07-06:
Do you have any BMS connected to your inverter?
How many BMS?

flopp999, 2025-07-06:
Is the entity enabled or disabled?
Mine is disabled and it looks like this.
[image]

flopp999, 2025-07-25:
@maartentamboer still error?

maartentamboer, 2025-07-25:
Changing to PV indeed resolves the issue:
```python
elif seriesnumber.startswith('AL1'):  invertertype = PV | GEN4 | X1 # Hybrid TL-XH 2.5kW - 6kW (MIN)
```