wills106/homeassistant-solax-modbus #1366: [Bug]: Enabling any modbus RC control mode, switches off inverter PV - HA 2025.5.0 issue?

jefft4, 2025-05-10:
### Describe the bug

This morning, my inverter's output suddenly dropped from 4kW to zero. I've identified through testing, that this now happens every time I set the modbus RC mode to anything other than 'disabled'. Power control with my normal parameters, zero output. Self-use, zero output.
Looking at the inverter statistics, I see that when the output is zero, the PV input voltages are high (320V vs normal ~290V in sunshine) and the current is zero. That looks like something has told the inverter 'disable PV input'.
As soon as I set the mode to 'disabled', returning the inverter to its Solax default 'self-use' operation, the PV power comes back up to full.

There are several repeated errors in the HA log from the integration and/or pymodbus. I'm not sure which might be relevant to this.  I'll switch on debug logging later today and see what I get from it.

### Integration Version

2025.04.4

### Homeassistant core version

2025.5.0

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H4502A

### Inverter firmware versions

DSP v1.46 ARM v1.43

### Connection Method

Solax wi-fi

### Dongle firmware

?

### Detailed Error Log

First repeated error:
```
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:583
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 11:23:54 (2 occurrences)
Last logged: 11:26:54

Modbus: ERROR: expected transaction 5 but got 4, IGNORING.
Modbus: ERROR: expected transaction 10 but got 9, IGNORING.
```

Second repeated error:
```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:112
First occurred: 11:23:55 (2 occurrences)
Last logged: 11:26:55

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
```

Also:
```
Logger: pymodbus.logging
Source: runner.py:154
First occurred: 11:24:07 (3 occurrences)
Last logged: 11:24:23

Exception response 144 / 0
```

Paiklabs, 2025-05-10:
I have simillar problem.
Setting "Export Control" and "Power limit" changes to zero by itself. 
I have turned off automation and external control and the error still occurs. 
The operating mode automatically falls into "Enabled SOC Target Control" mode, but this is not required.

I found that "Modbus Power Control" set to state "Enabled SOC Target Control" by itself every hour.
Export Control sometimes goes to 300 000 W when error occurs.

[image]

[image]

jefft4, 2025-05-10:
Debug log below. I've removed a few lines from other integrations/template sensors grumbling. Just before 15:45, I switched modbus RC from disabled to 'enabled self-use' and as before, the PV seemed to be shut off completely. The 'exception response' lines look interesting, along with the modbus 'wrong transaction' errors from the HA log this morning.

[image]

Looking at some of the sensors:
ActivePower Actual = 0
ActivePower Target = -550-ish varying
AP lower -5500 ; upper 5500
House load 688W
Measured power 0~60W varying
PV voltages are there but current is bouncing all over the place, 0A one minute and 5A the next. It should be around 5A at present. Mostly the current is <1A on each string, but it's not persistently zero this time.

Once I set it back to disabled, the current is 4~5A each string and we have 2.6kW of PV.

```
2025-05-10 15:41:40.720 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 22 but got 21, IGNORING.
2025-05-10 15:41:40.806 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-05-10 15:42:06.763 WARNING (MainThread) [custom_components.localtuya.common] [bff...hes] Failed to connect to 192.168.10.34: [Errno 111] Connect call failed ('192.168.10.34', 6668)
2025-05-10 15:43:06.509 WARNING (MainThread) [custom_components.localtuya.common] [bff...hes] Failed to connect to 192.168.10.34: [Errno 111] Connect call failed ('192.168.10.34', 6668)
2025-05-10 15:44:06.171 WARNING (MainThread) [custom_components.localtuya.common] [bff...hes] Failed to connect to 192.168.10.34: [Errno 111] Connect call failed ('192.168.10.34', 6668)
2025-05-10 15:44:10.672 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 8 but got 7, IGNORING.
2025-05-10 15:44:10.817 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-05-10 15:45:05.726 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-05-10 15:45:06.360 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2025-05-10 15:45:09.185 INFO (MainThread) [custom_components.solax_modbus.select] *** local data written remotecontrol_power_control: 110
2025-05-10 15:45:12.005 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] Evaluated remotecontrol_trigger: corrected/clamped values: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', -647), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 30.0)]
2025-05-10 15:45:12.005 INFO (MainThread) [custom_components.solax_modbus.button] writing SolaX button register 124 value [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', -647), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 30.0)]
2025-05-10 15:45:12.006 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0x7c: [1, 1, 64889, 65535, 0, 0, 30]
2025-05-10 15:45:12.103 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-10 15:45:15.726 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-05-10 15:45:16.271 INFO (SyncWorker_6) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'remotecontrol_active_power': 5500.0, 'remotecontrol_reactive_power': 0, 'remotecontrol_duration': 30.0, 'remotecontrol_autorepeat_duration': 660.0, 'remotecontrol_import_limit': 10000.0, 'config_export_control_limit_readscale': None, 'config_max_export': 15000, 'remotecontrol_power_control': 'Enabled Self Use', 'remotecontrol_set_type': 'Set'}
[... cut ...] trol_reactive_power', 0), ('remotecontrol_duration', 30.0)]
2025-05-10 15:45:16.272 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0x7c: [1, 1, 64980, 65535, 0, 0, 30]
2025-05-10 15:45:16.368 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-10 15:45:16.376 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2025-05-10 15:45:25.727 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter

```

PiotrRaszkowski, 2025-05-26:
I receive also Exception response 144 / 0.
Is it related?
```
2025-05-26 14:02:36.579 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:02:51.582 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:03:06.545 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:03:30.038 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:03:38.354 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:03:51.498 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:04:06.521 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:04:21.524 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:04:42.566 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:04:51.519 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-05-26 14:04:53.471 DEBUG (MainThread) [custom_components.solax_modbus.plugin_sofar] check_battery_on_end: 0x0 0x0
2025-05-26 14:04:53.471 DEBUG (MainThread) [custom_components.solax_modbus.plugin_sofar] batt pack serial: 
2025-05-26 14:04:53.474 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2025-05-26 14:05:00.220 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
```

wonko, 2025-06-14:
I seem to be having the same problem (or something along the same lines). Upgraded to latest stable HA and Solax Modbus latest release last week, and when the system tried to dump some battery power to the grid this morning, everything started acting wierdly. I'll include a good part of the log, with some python dumps in them, but essentially it just loses connection to the inverter. 

My setup is a G4 Hybrid X3 with pocketwifi. Has been working flawless for a couple of years now, and I've been using the remote control through HA scripting a lot (dynamic power tariff). Scripting wasn't touched during the upgrade, and otherwise the upgrade hasn't given any issues in other plugins/scripts/..., both with and besides the Solax entities/devices.

I have 

```
homeassistant  | 2025-06-14 07:51:17.249 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('192.168.1.52', 502)
homeassistant  | 2025-06-14 07:51:17.803 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.52:502
```

and

```
homeassistant  | 2025-06-14 07:56:41.716 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request
homeassistant  | 2025-06-14 07:56:41.716 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 29 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
```

I believe the issue lives in the inverter itself, as It seems that the command overwhelms the pocketwifi / internal modbus server somehow, as from that moment onwards, the "app" (which just pushes these metrics to some cloud service) also looses it's data.

Could be something that overwhelms the internal system - did something change in the frequency of polling, or are connection re-established over and over ...

closed 2025-09-16 by github-actions[bot]