wills106/homeassistant-solax-modbus #1383: [Bug]: 'growatt' - 'sph 10000tl3 bh-up' Exception response 131 / 0

MEGABUSTER023, 2025-05-24:
### Describe the bug

Hi,
I just bought a Growatt Data Bridge from SolarSmart.dk. I've set it up and connected it to my Home Assistant.
After that, I tried to create an automation to charge the batteries from the grid. But every time I try to activate AC Charge in Home Assistant, nothing happens.

When I check the ShinePhone app (the app we use for Growatt), I can see that AC Charge hasn’t been activated. And when I try to change the setting Battery First Time from Grid, it just keeps resetting the time to 00:00.

Does anyone know why this is happening?

### Integration Version

2025.04.4

### Homeassistant core version

2025.5.3

### Inverter brand

growatt

### Plugin used

plugin_growatt.py

### Serial prefix

no idea

### Connection Method

rs485 ESP32 Modbus Gateway

### Detailed Error Log

Logger: pymodbus.logging
Source: runner.py:154
First occurred: 12:52:02 (20 occurrences)
Last logged: 13:11:42

Exception response 131 / 0
Exception response 132 / 0

### Additional context

I used the integration from hacs solax inverter modbus the website i bought my bridge from is https://solarsmart.dk/vare/growatt-data-bridge-home-assistant

flopp999, 2025-07-25:
@MEGABUSTER023 can you please update to latest versions and see if the problem is solved?

MEGABUSTER023, 2025-07-25:
@flopp999 It worked yes 🙌 . Btw how would I go about making a automation that can make the battery charge from the grid? I want to make a automation that depending on electricity prices makes the battery charge from the grid

closed 2025-08-31 by wills106