wills106/homeassistant-solax-modbus #1384: [FR]: Bias Power entity for SolaX inverter type

xmartins2, 2025-05-24:
### Description

Bias Power entity is not implemented for inverter type of solax.

### Suggested Solution

Implement Bias Power entity for inverter type of solax.

### Additional Context

Pgrid Bias entity is implemented.
Solax X3-Hybrid-G4

xmartins2, 2025-05-29:
I have found user description of Bias Power in this user manual: https://www.solaxpower.com/uploads/file/x3-hybrid-g4-series-user-manual-v7-en.pdf
In advanced settings of my inverter I can see the following about Bias: 

[image]

wills106, 2025-06-10:
> I have found user description of Bias Power in this user manual:

Unfortunately I need the Modbus Registers which perform the same function.

xmartins2, 2025-06-11:
Unfortunately I do not know the Modbus Registers of Bias Power.

cscholz, 2025-07-27:
> I'was able to identify the read-only register for "pgrid bias power" which is **0x02A7**
> 
> ```
> $ mbpoll -m tcp -a 0 -t 4 -r 679 -c 1 xx.yy.zz.ip
> ```
> 
> Wrote a script to read all register. Changed the bias power value and read all register again. Done this multiple times and make sure the identified is the right one.
> 
> Write register seems to be another one as writing fails
> 
> ```
> $ mbpoll -m tcp -a 0 -t 4 -r 679  -1 xx.yy.zz.ip 1
> [...]
> Protocol configuration: Modbus TCP
> Slave configuration...: address = [0]
>                         start reference = 679, count = 1
> Communication.........: xx.yy.zz.ip, port 502, t/o 1.00 s, poll rate 1000 ms
> Data type.............: 16-bit register, output (holding) register table
> 
> Write output (holding) register failed: Illegal data value
> ```
> 
> Contacted Solax and requested a confirmation and also an information which register is the one to write.
> 
> Update follows... I hope

@wills106: Can you check if you can at least add a sensor for the moment to show the "bgrid bias power"?
I've not yet received an answer from qcell/solax