wills106/homeassistant-solax-modbus #1390: [Bug]: 'Inverter Growatt' - 'Inverter MID25KTL3XH' - 'Missing PV 3'

banjohb, 2025-05-27:
### Describe the bug

I’m not seeing any values for my 3rd MPPT tracker (PV 3) in the system. Could there be a reason for this, and is there a way to fix it?

### Integration Version

2025.04.4

### Homeassistant core version

2025.5.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

DN1.0/DNBA025101

### Connection Method

RS485 Waveshare USB

### Detailed Error Log

Logs are ok

### Additional context

All other entities look good and are working fine.

banjohb, 2025-05-27:
I fixed it :)

https://github.com/wills106/homeassistant-solax-modbus/issues/1269#issuecomment-2911245914

wills106, 2025-05-28:
Do you want to submit a PR with the change?

banjohb, 2025-07-25:
To see the data from your third PV string, you need to adjust the file according to the instructions here:

https://github.com/wills106/homeassistant-solax-modbus/issues/1269#issuecomment-2911245914

However, it gets overwritten every time SolaX updates.

Could the developer maybe include this permanently in the integration?

banjohb, 2025-08-03:
Please fix it. Here is the solution:

https://github.com/wills106/homeassistant-solax-modbus/issues/1519

banjohb, 2025-08-03:
> @banjohb you can upload a PR then the developer can add it.

I don't know how 🥸 here is the description: https://github.com/wills106/homeassistant-solax-modbus/issues/1519

closed 2025-09-17 by github-actions[bot]