wills106/homeassistant-solax-modbus #1393: [Bug]: Entities are not updated for several minutes, HA shows old values

janouskovec, 2025-05-28:
### Describe the bug

I see in log following error:
2025-05-28 11:10:43.047 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 3850 but got 3849, IGNORING.
2025-05-28 11:10:43.161 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 192, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state

When error occurs, HA show old values for entities (entity values ​​are not updated). After a few minutes, it starts working normally again.

### Integration Version

2025.04.4

### Homeassistant core version

2025.1.0

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM 1.47, DSP 1.50

### Connection Method

Solax Pocket Wi-Fi 3.0

### Dongle firmware

3.019.01

### Detailed Error Log

2025-05-28 11:10:43.047 WARNING (MainThread) [custom_components.solax_modbus] Modbus: ERROR: expected transaction 3850 but got 3849, IGNORING.
2025-05-28 11:10:43.161 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 192, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state

ruisilva93, 2025-05-30:
Same. Related with #1400 ??

janouskovec, 2025-06-01:
I tried to upgrade HA core to version 2025.5.3 and integration to version 2025.05.3, but I still see same errors in the log.

closed 2025-09-13 by github-actions[bot]