wills106/homeassistant-solax-modbus #1401: [Bug]: 'Growatt' - 'MOD6000TL-XH' No connection after updating to 2025.05.2

rulez86, 2025-05-31:
### Describe the bug

After updating 2025.05.1 to 2025.05.2 i get no data from my inverter.

I'm using two inverters. Growatt MIC3300TL-X and MOD6000TL-XH. MIC is working fine, MOD is no longer working after the update, means i get no data. See error message above. Both are connected by separate USB-RS485 Adapter.

### Integration Version

2025.05.2

### Homeassistant core version

2025.5.3

### Inverter brand

GROWATT

### Plugin used

plugin_growatt.py

### Serial prefix

DNAA045100

### Connection Method

RS485 over USB

### Detailed Error Log

```
2025-05-31 09:38:03.149 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x13bb bms_2_monitoring_version
2025-05-31 09:38:03.177 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x13bb bms_2_monitoring_version
2025-05-31 09:38:18.588 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 09:38:18.614 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 09:38:18.640 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 09:38:18.666 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 09:38:18.692 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 09:38:18.719 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 09:38:20.903 ERROR (MainThread) [custom_components.solax_modbus] ModbusIOError: Modbus Error: [Input/Output] Modbus value 20817 not found in scale mapping for bms_2_module_1_status
```

rulez86, 2025-05-31:
> [@user86000](https://github.com/user86000) can you create a separate issue, as you are probably getting a similar issue. But on completely different registers.
> 
> [@rulez86](https://github.com/rulez86) do you get the same issue on [2025.05.3](https://github.com/wills106/homeassistant-solax-modbus/tree/2025.05.3)?

Hey @wills106 unfortunately same issue with 2025.05.3:

```
2025-05-31 20:11:10.783 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 20:11:10.809 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 20:11:10.835 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 20:11:10.861 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 20:11:10.887 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 20:11:10.913 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-31 20:11:13.154 ERROR (MainThread) [custom_components.solax_modbus] ModbusIOError: Modbus Error: [Input/Output] Modbus value 20817 not found in scale mapping for bms_2_module_1_status
2025-05-31 20:11:25.783 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
```

KasperEdw, 2025-05-31:
Im using Growatt SPH3600 - I also lost data on update.

derjape, 2025-06-01:
From my MOD 10000tl3-xh I also do not receive any data since 2025.05.3

Two error messages in logfile:
1st error

 Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:725
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 07:40:07 (13 Vorkommnisse)
Zuletzt protokolliert: 07:50:02

ModbusIOError: Modbus Error: [Input/Output] Modbus value 20817 not found in scale mapping for bms_2_module_1_status

2nd error
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus.sensor
Quelle: custom_components/solax_modbus/sensor.py:256
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 07:39:59 (1 Vorkommnis)
Zuletzt protokolliert: 07:39:59

holding register already used: 0x13bb bms_2_monitoring_version

MkVsM, 2025-06-01:
The same problem with Solax X3 Hybrid G1/G2. It work only 2025.05.1.

TonyMathiesen, 2025-06-06:
Hi, could you try with this: https://pastebin.com/FtHGL3bc 
I believe that it will solve this issue. 
In @flopp999's large PR #1338 there is a lot of improvements. Also a way to handle the scaling of these registers. It uses a "value_function_module_status" where it can handle not listed values. I created a PR #1420 with a small part of @flopp999's PR, as I believe that can quickly solve this issue.

closed 2025-06-10 by wills106

rulez86, 2025-06-15:
@wills106 
sorry for late response. Works fine for me. Thanks a lot.

commit 08c18b52 (2025-06-10): Merge pull request #1420 from TonyMathiesen/patch-4

Growatt - Fix #1401, #1405, #1415