wills106/homeassistant-solax-modbus #1404: [Bug]: Inverter connection lost after upgrade to 2025.05.3

ezfrag2021, 2025-06-01:
### Describe the bug

All entities unavailable after update to latest version.

### Integration Version

2025.05.3

### Homeassistant core version

2025.5.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H3DE05

### Inverter firmware versions

Unknown

### Connection Method

Ethernet

### Dongle firmware

2.034.03

### Detailed Error Log

This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:725
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 07:35:19 (6 occurrences)
Last logged: 07:47:34

ModbusIOError: Modbus Error: [Input/Output] Modbus value 5 not found in scale mapping for meter_1_direction

ezfrag2021, 2025-06-01:
Rolling back the integration version to 2015.05.2 or 2025.05.01 did not fix the issue. Problem may be due to upgrade of HA Core.

godfreym29, 2025-06-01:
I'm also seeing this

**Detailed Error Log**
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:725
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 10:10:59 (8 occurrences)
Last logged: 10:28:14

ModbusIOError: Modbus Error: [Input/Output] Modbus value 48 not found in scale mapping for backflow_power_switch

also getting a lot of these errors in the log

```
2025-06-01 10:33:48.986 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.solis_battery_charge_current for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 979, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 259, in async_internal_added_to_hass
    raise HomeAssistantError(
        f"Entity {self.entity_id} cannot be added as the entity category is set to config"
    )
homeassistant.exceptions.HomeAssistantError: Entity sensor.solis_battery_charge_current cannot be added as the entity category is set to config
2025-06-01 10:33:48.987 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.solis_battery_discharge_current for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 979, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 259, in async_internal_added_to_hass
    raise HomeAssistantError(
        f"Entity {self.entity_id} cannot be added as the entity category is set to config"
    )
homeassistant.exceptions.HomeAssistantError: Entity sensor.solis_battery_discharge_current cannot be added as the entity category is set to config
2025-06-01 10:33:48.988 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.solis_timed_charge_current for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 979, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 259, in async_internal_added_to_hass
    raise HomeAssistantError(
        f"Entity {self.entity_id} cannot be added as the entity category is set to config"
    )
homeassistant.exceptions.HomeAssistantError: Entity sensor.solis_timed_charge_current cannot be added as the entity category is set to config
2025-06-01 10:33:48.989 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.solis_timed_discharge_current for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 657, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 979, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 259, in async_internal_added_to_hass
    raise HomeAssistantError(
        f"Entity {self.entity_id} cannot be added as the entity category is set to config"
    )
homeassistant.exceptions.HomeAssistantError: Entity sensor.solis_timed_discharge_current cannot be added as the entity category is set to config
```

decibelhuis, 2025-06-01:
The same problem with solis rhi-3p10k-hves-5g via LAN. It work with 2025.05.01

felixhix, 2025-06-02:
Have a similar issue since Update to 2025.5.3 but with Growatt Inverter.
```
Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:725
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 1. Juni 2025 um 22:00:13 (256 Vorkommnisse)
Zuletzt protokolliert: 08:36:12

ModbusIOError: Modbus Error: [Input/Output] Modbus value 5 not found in scale mapping for grid_first_time_6
```

TommasoPino, 2025-06-06:
I think it is the same error. It happens after a restart of hassio

Registratore: custom_components.solax_modbus.plugin_sofar
Fonte: custom_components/solax_modbus/plugin_sofar.py:4161
Integrazione: SolaX Inverter Modbus (documentazione, problemi)
Prima occorrenza: 14:23:22 (1 occorrenza)
Ultimo accesso: 14:23:22

unrecognized ZCS2 inverter type - serial number : unknown

felixhix, 2025-08-28:
I can confirm that entities will appear normally again with version 2025.08.7.

felixhix, 2025-08-30:
Unfortunately I have to take back my statement, some entities have ‘come back’, but some values are still missing compared to the ‘stable’ version.

These are in particular:
- Today's Solar Energy
- Grid Power L2+L3
- AC Power to User Total

infradom, 2025-08-30:
@felixhix : unsufficient info, no model info, no debug logfile, no HA verson. 
ignoring ...

felixhix, 2025-08-30:
Sorry for that 😄 :
- HA: 2025.8.3
- Inverter: Growatt SPH 10k TL3-BH UP
- Solax: 2025.08.7

Solax Log-Level is currently at INFO. Any advice to switch it to DEBUG?

felixhix, 2025-08-30:
Yes, see log:
`2025-08-30 11:58:20.495 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: trying to determine inverter type
2025-08-30 11:58:20.585 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt 0x9 firmware number before potential swap: YA1.0 ZDAA`.

This matches with the informations from the Growatt webportal:

[image]

infradom, 2025-08-30:
Please report your problems in a dedicated thread
(most of them have already been discussed)

infradom, 2025-08-30:
I have explained the disabled entities problem multiple times in other isssue threads, The quickfix is to enable the underlying register entities

felixhix, 2025-08-30:
Thanks for the clarification. I have activated it and the value is now calculated. Just a side note: The SPH only has 2 MPP trackers. Isn't it actually the case that the SPH has its own register for the Today ‘Total’? I may be wrong, though.
Growatt Documentation: https://community.symcon.de/uploads/short-url/jFo2xrUpTus03y9fWeYrp0Dbfng.pdf