wills106/homeassistant-solax-modbus #1412: [Bug]: Upper SoC can no longer be adjusted

richie241, 2025-06-02:
### Describe the bug

Hello everyone,

The function was still working some time ago, but now I have the following problem:
On a Solax X3 Ultra with 2 batteries, the 'Battery Charge Upper SOC' can no longer be controlled or is not transmitted to the inverter. The same applies to the ‘Battery Charge Max Current’ and ‘Battery Discharge Max Current’ values

The reason is probably that the inverter now looks at the values for BAT1 and BAT2. See screenshot attached.

Likewise, no temperature is output for BAT2. This had also worked before.

Translated with DeepL.com (free version)

### Integration Version

2025.05.3

### Homeassistant core version

2025.5.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H3BF25

### Inverter firmware versions

ARM: 019.02-000.10 DSP: 018.03

### Connection Method

Solax Pocket LAN/Wifi 3.0

### Dongle firmware

1.005.04

### Additional context

[image]
[image]

Krawuzzi, 2025-06-13:
For me these values can be changed via home assistant, but only for a limited time (about 1 day or so) after logging into the solax cloud and going to the advanced settings.
After a while the changes via home assistant are no longer working, then I need to log into the cloud again.

I have the Solax X3-Ultra-25kW.
I think it has something to do with the authentication that expires after a while.

richie241, 2025-06-15:
> For me these values can be changed via home assistant, but only for a limited time (about 1 day or so) after logging into the solax cloud and going to the advanced settings. After a while the [...]

Do you have 1 or 2 Batterys connected?

infradom, 2025-09-15:
I personally ignore bug reports without full debug log (as described in the genaral FAQ)
This is the reason why I did not look into this. It may however be relatively simple to solve. reopening, but a full debug log is an incentive for me to look into this ... I like logs ..(even without error messages)