wills106/homeassistant-solax-modbus #1413: [Bug]: Solax X3-Hybrid-G4  Sensor "House load" and "house load alt" wrong value

giovannirat, 2025-06-03:
### Describe the bug

The value of "Solax inverter house load" and Solax inverter house load alt" are most of the time not correct and totally different from the value shown in the Solax App (even considering the fact that due to time differences are some deviations). See also attachment (the bottom value on the GeekMagic is the "house load alt" value. Sometimes the value is negative, sometimes positive, which makes no sense. 
There is only one situation when the value seems correct: When the battery is supplying the house (no grid, no solar modules) the value seems correct (Battery output= house load).

[image]
[image]
[image]

### Integration Version

2025.05.3

### Homeassistant core version

2025.5.3

### Inverter brand

Solax X3-Hybrid Gen4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP v1.48 ARM v1.45

### Connection Method

Elfin EW11A Wireless Modbus Adapter

### Dongle firmware

1.41

giovannirat, 2025-06-04:
Attached a graph over the last week for "Solax Inverter House Load Alt"

[image]

giovannirat, 2025-06-04:
Closed accidentaly

TCWORLD, 2025-06-08:
Just had a SolaX X1-IES installed and got it connected up to HA. Pretty sure I'm having the same issue.

This is the initial data from the inverter:
[image]

House load is showing as negative, which I believe is something to do with the "Inverter Power" reading as zero. As a result the calculation of "Inverter Power" - "Measured Power" is giving the wrong result.
At the moment the battery is not charging, so the difference between "PV Power Total" and "Measured Power" gives the correct house load, though I guess that may change when the battery is being charged.

infradom, 2025-07-16:
Is there something worth mentioning regarding your battery string(s)? Do you have multiple battery strings?

itrankolov, 2025-07-21:
Hello,

I have the same issue. X3-Ultra-30kW, 1 battery string, 2 PV strings. No power meter 2 (CTs attached to inverter).

giovannirat, 2025-09-28:
This issue is still present. Wrong value for house load

rosenrot00, 2025-09-30:
Is the house load off all the time by a certain factor?

Could you tell me what your fast polling rate is when you see this error?

giovannirat, 2025-09-30:
I did not see any factor, but as I said in my initial statement "There is only one situation when the value seems correct: When the battery is supplying the house (no grid, no solar modules) the value seems correct (Battery output= house load)."

Polling rate is 10 sec.

infradom, 2025-10-03:
That is meant for an optional generator. So most people dont have a generator connected, so 0 is ok

fxstein, 2025-11-16:
@giovannirat Sorry - can you explain what you are seeing - and what you think is not correct? I cannot tell from the chart you posted if this is good or bad or what is wrong with it. Thank you!

giovannirat, 2025-11-17:
Sorry for the late response...

First of all I think the "house load" and "house load alt" should not become a negative value (in the chart you see a nearly -3000 value at 12.00o'clock).

Furthermore the it seems that the value  of "house load" and "house load alt" sometimes do not correspond to the value shown by the Solax APP  (i will try to find examples)

slesinger, 2025-12-30:
I wonder if this is related. I use solax X3 G4 connected via TCP/RS485 converter [USR-W610](https://www.pusr.com/products/rs232-rs485-to-wifi-converters-usr-w610.html)
Battery SoC should be within 0-100%, obviously but it jitters up to 18000%
It was working absolutely fine 2 months ago. Then, I tried to switch to the Solax internal integration and then came back to this HACS one. I believe, I have not done other changes than upgraded HASS and this package in the meantime.
Current value of 83% seen on the screenshot is true.
[image]