wills106/homeassistant-solax-modbus #1417: [FR]: 'Complement time slots for Growatt MOD inverter to 9 in total'

MuadDib007, 2025-06-05:
### Description

I would like to see the number of Time Slots for scheduled charge/discharge (Which is currently limited at 3) to be increased to 9 in total.

I know this is available in the Growatt MOD Hybrid inverter series, e.g. MOD10KTL3/XH(BP)

### Suggested Solution

The slots could be implemented to work the same as the already available slots 1-3.

### Alternatives

You could re-write a timeslot when it is out of use, but that leads to more wear and tear of the EEPROM of the inverter.
Therefore it would be convenient to have 9 time slots available.

### Additional Context

Registers can be found in this document:
https://github.com/johanmeijer/grott/blob/master/documentatie/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-24-English-new.pdf

Specifically it should be these registers:
Time4: 3044/3045
Time5: 3050/3051
Time6: 3052/3053
Time7: 3054/3055
Time8: 3056/3057
Time9: 3058/3059

flopp999, 2025-07-05:
You can send a PR for this, then it would be much raider for the maintainers to add it

flopp999, 2025-07-06:
I can add it if you confirm that have not started to do it?

MuadDib007, 2025-07-06:
I have not started on it.
Thank you for helping with this request.

flopp999, 2025-07-06:
Ok
Then I will do it.
No problem

flopp999, 2025-07-06:
I have code that works, but I have some changes in my setup, that is not yet implemented to the release here at GitHub.
I will upload a PR when some other changes has been added.
If you want to use my code you can download it here https://raw.githubusercontent.com/flopp999/homeassistant-solax-modbus/refs/heads/main/custom_components/solax_modbus/plugin_growatt.py

But please make a backup of the code you are using now, before you overwrite with my code

flopp999, 2025-07-07:
@MuadDib007 I have now uploaded a PR for this
the code that I uploaded you can find here
https://raw.githubusercontent.com/wills106/homeassistant-solax-modbus/ab31b064eeaa379db8a4b0ed47b75fe398feff1a/custom_components/solax_modbus/plugin_growatt.py

flopp999, 2025-07-16:
@MuadDib007 all 9 timers are available in the latest release of SolaX, when you have tested it and it works, please close this issue

MuadDib007, 2025-07-17:
I have tried but fail to change te timer settings.
Output from Debug logging is as follows:
2025-07-17 13:36:03.875 INFO (MainThread) [custom_components.solax_modbus.button] writing G10K button register 3038 value [('_uint16', 40960), ('_uint16', 5947), ('_uint16', 16916), ('_uint16', 552), ('_uint16', 0), ('_uint16', 0), ('_uint16', 0), ('_uint16', 0), ('_uint16', 0), ('_uint16', 0), ('_uint16', 16414), ('_uint16', 30), ('_uint16', 8207), ('_uint16', 15), ('_uint16', 8192), ('_uint16', 0), ('_uint16', 17674), ('_uint16', 1536)]
2025-07-17 13:36:03.875 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0xbde: [40960, 5947, 16916, 552, 0, 0, 0, 0, 0, 0, 16414, 30, 8207, 15, 8192, 0, 17674, 1536]

Could the issue be caused by trying to write 20 consecutive registers while there is a little jump in register number between time4 and time5?

flopp999, 2025-07-17:
I tried again and it seems that I can write time 3 and read time, without problem.
I will take a look again and see if I missed something.

MuadDib007, 2025-07-17:
It seems to be reading correctly (Changes from ShinePhone become visible in HA) but changes I try to make from HA only seem to get written to the inverter rarely.
I haven't found a pattern yet.
To prevent the inverter from being over-asked I did disable many entities from the integration to see if that would fix the issue.
So far I have only been able to change the setting for Time2 once.

flopp999, 2025-07-17:
I will go for one button to update each timer and one button to clear each timer.
Lets say you would like to use timer 7 and change the settings for that, before you can press update 5-9, you need to check 5,6,8,9 before you press Update 5-9, otherwise there could be a timer that will not fit your chose

closed 2025-07-27 by MuadDib007