wills106/homeassistant-solax-modbus #1418: [Bug]: Solax X3 Hybrid G1/2 no values

Voodoomoped, 2025-06-05:
### Describe the bug

Updated to current version of HA with 2025.05.03, Inverter is detected properly but entities show no values even after several restarts. Port 502 is open, IP correct. 
Would be great to get help here. 
Thanks in advance, Daniel

### Integration Version

2025.05.3

### Homeassistant core version

2025.5.3

### Inverter brand

Solax X3 Hybrid

### Plugin used

plugin_solax.py

### Serial prefix

H3UE

### Inverter firmware versions

Arm 4.43 -1.05

### Connection Method

Modbus TCP via FRITZ!Box (has been working with older version of HA)

### Dongle firmware

Not used here

### Detailed Error Log

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solax_modbus
Quelle: custom_components/solax_modbus/__init__.py:725
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 21:54:28 (6 Vorkommnisse)
Zuletzt protokolliert: 22:06:43

ModbusIOError: Modbus Error: [Input/Output] Modbus value 6 not found in scale mapping for meter_2_direction

RainerHemerka, 2025-06-07:
Hello....
I have the same problem as well with my Solax X3-Hybrid-10kW Inverter.
After updating to version 2025.05.03 of the integration, all sensors became unavailable.
I had the same problem with previous version 2025.05.02 which i skipped.
My connection method is Modbus TCP.
One Inverter Works, but the other via modbus Proxy not.

After rolling back to the previous version, all sensors an inverters work again.

chrischeu, 2025-06-07:
Same issue on my setup with Solax X3 Hybrid using current version 2025.05.3.
My connection method is Modbus TCP.
After downgrading to version 2025.05.1 all entities work as expected and the of the values of the entities are available.

sp0ok3r, 2025-06-09:
does Solax X3 Hybrid G1/2 got modbus? or you connected waveshare modbus ethernet?