wills106/homeassistant-solax-modbus #1423: [Bug]: 'Growatt' - 'MOD8000TL3-X' Several values change unexpectedly when the inverter goes offline in the evening

MuadDib007, 2025-06-10:
### Describe the bug

Many values are affected but the following should at least change behaviour in my opinion:

Current behaviour:
Output Power --> Keeps value of last reading
Today's Solar Energy --> Set to 0 when inverter goes offline when PV input is too little.
Today's Power Generation --> Keeps value of last reading, set to 0 when inverter comes online when PV input is starting.
Today's PV1 Solar Energy --> Keeps value of last reading, set to 0 when inverter comes online when PV input is starting.
Today's PV2 Solar Energy --> Keeps value of last reading, set to 0 when inverter comes online when PV input is starting.
PV Power Total --> Keeps value of last reading
PV Voltage 1 --> Keeps value of last reading
PV Voltage 2 --> Keeps value of last reading
Grid Current L1 to L3 --> Keeps value of last reading
Grid Power L1 to L3 --> Keeps value of last reading

Expected behaviour:
Output Power --> Set to 0 when inverter is offline
Today's Solar Energy --> Keep latest value when connection is lost. Reset to 0 at 00:00
Today's Power Generation --> Reset to 0 at 00:00
Today's PV1 Solar Energy --> Reset to 0 at 00:00
Today's PV2 Solar Energy --> Reset to 0 at 00:00
PV Power Total --> Set to 0 when inverter is offline
PV Voltage 1 --> Set to 0 when inverter is offline
PV Voltage 2 --> Set to 0 when inverter is offline
Grid Current L1 to L3 --> Set to 0 when inverter is offline (Or Unknown?)
Grid Power L1 to L3 --> Set to 0 when inverter is offline (Or Unknown?)

Maybe this list is not extensive but this is what I have found so far.

### Integration Version

2025.06.01

### Homeassistant core version

2025.5.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

RHHDCK

### Connection Method

Modbus TCP to ShineWiLan-X2

MuadDib007, 2025-07-05:
Just tested to restart HA when the inverter is offline. All values are unavailable.

I am unable to start the inverter as there is no solar power.

In the morning when the inverter starts again the values are restored.

What behaviour should we expect from the integration in regards to inverters going offline because of lack of solar energy?

flopp999, 2025-07-05:
If the inverter goes offline/sleep it will not reply to the Modbus questions that HA is sending, so then I assume it will show unavailable or maybe the last received value.
If you restart HA during this, I think it will be unknown/unavailable. Same
Result as you put wrong address or IP for the inverter.
HA cannot make a connection to the inverter.