wills106/homeassistant-solax-modbus #1446: [Bug]: 'Growatt' - 'SPH 10000TL3 BH-UP' curios Data received

alfiriel, 2025-07-02:
### Describe the bug

(I do not know what "Plugin used" means. I choose "growatt" for inverter type in the configuration settings)
**Working: 2025.01.8**

With the latest Version (**2025.06.2**) the integration is not working correctly with my inverter Growatt SPH 10000TL3 BH-UP:

The PV system's production supposedly increases at night. At some point, the value then shows 0 W, and data only comes back when I restart the entire integration.

I searched the issues for my error and found the following issue: https://github.com/wills106/homeassistant-solax-modbus/issues/1222

I then installed the version 2025.01.8 mentioned in the issue. Everything works as it should.

Something happened between this version and the current one, so the inverter with firmware 050510 is no longer supported.

Can you help with this?

### Integration Version

2025.06.2

### Homeassistant core version

2025.6.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

050510

### Connection Method

RS485-USB Waveshare FT232 Chip

flopp999, 2025-07-05:
Great. Yes, logs would be good to get. If you can copy the error log for SolaX when it is not working.
Maybe the inverter is sleeping and then you cannot communicate with it.

alfiriel, 2025-07-05:
There are no error logs. Just wrong data receiving.

In the Screenshots you can see in the morning the PV Power is at 0 W. This cannot be because the sun ist already shining way enough.
[image]

After restarting the Integration, there are the correct values:
[image]

In the night, my PV definitely does not produce power:

[image]

flopp999, 2025-07-05:
Ok, yeah that seems very strange that your PV shows power during night.
Correct time and location in HA?
What I know, HA is only getting live data, so no historical data. This means that the inverter really reply with those values, but a restart will give you the correct values, which shows that there must be something else.
You have a USN dongle to Modbus RTU, directly connected to inverter?
Do you have any other Modbus connection to HA or inverter?

alfiriel, 2025-07-05:
Even more curios that it was working like a charm with Plugin Version 2025.01.8 for 4 days now.

Yes correct location and time in HA.

I use a Waveshare Industrial USB to RS485 Converter with FT232RNL chip. Connected via Cat 7 cable to the Inverter.

No other modbus connection in HA 

I have an Shine WiFi Stick connected to the Inverter which ist flashed with https://github.com/OpenInverterGateway/OpenInverterGateway

Side Note: configured HA and the solix intergration also for my parents in their house. They have Growatt SPH 5000TL3 BH-UP with the Same Firmware (050510) and everything is working Fine with the latest solix Version.

flopp999, 2025-07-05:
Ahh so with older version of SolaX it works.
I don't know what difference there is between latest version and that one working, but I can have a Quick Look and compare them tomorrow.
And you parents do not have custom firmware in ShineWifi stick?

alfiriel, 2025-07-05:
Sorry forgot to mention. They are using the original Shine stick with cloud connection.

I only have the modified stick in my Inverter because I wanted something cloud free tonstaet wih and set up HA only few weeks ago. So If it would help I could disconnect the stick at any time.

flopp999, 2025-07-06:
@alfiriel would it be possible for you to try with original firmware in stick?
Is there an advanced to use custom firmware in stick?
I read t quick about the CF and it mentioned Modbus, but I think the Modbus connection from HA is directly to the inverter, which means your stick should not cause any trouble.