wills106/homeassistant-solax-modbus #1469: Growatt MIC - Serial number becomes unavailable after update

thomas7475, 2025-07-15:
### Describe the bug

Hello, after updating from 2025.06.2 to version 2025.07.1, the "Serial Number" entity for my Growatt MIC inverter has become unavailable. I tracked it down to the change in plugin_growatt.py in the following function.

    GrowattModbusSensorEntityDescription(
        name = "Serial Number",
        key = "serialnumber",
        register = 3001,
        unit = REGISTER_STR,
        wordcount = 5,
        entity_registry_enabled_default = False,
        allowedtypes = GEN4,
        entity_category = EntityCategory.DIAGNOSTIC,
        icon = "mdi:information",
    ),

this function was changed to this 

    GrowattModbusSensorEntityDescription(
        name = "Serial Number",
        key = "serialnumber",
        register = 3001,
        unit = REGISTER_STR,
        wordcount = 5,
        entity_registry_enabled_default = False,
        allowedtypes = GEN4 | HYBRID,
        entity_category = EntityCategory.DIAGNOSTIC,
        icon = "mdi:information",
    ),

after removing the change | HYBRID it works again.

If i understand this correct this line  allowedtypes = GEN4 | HYBRID, checks if it is a gen 4 hybrid inverter and only then the entity is available in homeassistant please correct this logic. The serial number should be available for all GEN4 inverters, not just the HYBRID sub-type. Reverting the change or creating a separate definition for each type would resolve the issue.

### Integration Version

2025.07.1

### Homeassistant core version

2025.7.2

### Inverter brand

growatt

### Plugin used

plugin_growatt.py

### Serial prefix

GH1.0

### Connection Method

RS485-USB

kl1127, 2025-07-16:
Same here on my SPH3600

flopp999, 2025-07-17:
> Assuming it's in [2025.07.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.07.2) then?

Correct. In the PR you merge an hour ago is the fix for this :)

wills106, 2025-07-19:
@thomas7475 can this be closed down now?

thomas7475, 2025-07-20:
yes i closed it, it is working now.