wills106/homeassistant-solax-modbus #1470: [FR]: Growatt - Long term statistics

flopp999, 2025-07-16:
### Description

I noticed that some entities are not being stored in HA for long term.
Is there any entites that you want to be stored for long term?
Please add them in this issue and I can change so it will be long term.

### Suggested Solution

add state_class to the entity

flopp999, 2025-07-16:
> Why is this only Growatt related? Wouldn't it make sense to make it available to the entire integration as an opt-in option?

Because I have only knowledge about Growatt.

I have not check any other entities for this integration.

rosenrot00, 2025-07-16:
I have a Solax inverter with InfluxDB running on my end. So there is no strong need to have that in HA integrated but once an option is suggested how to implement it as an option I can help with the Solax integration.