wills106/homeassistant-solax-modbus #1473: [Bug]: Sofar - HYD 20KTL-3PH UnboundLocalError: cannot access local variable 'action' where it is not associated with a value

anton4, 2025-07-18:
### Describe the bug

After upgrading from 2025.06.02 to 2025.07.02 Passive update battery charge/discharge button (button.sofar_passive_mode_battery_charge_discharge) does not work anymore and giving this error in the logs - 
`2025-07-18 09:42:35.663 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140009461365024] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 250, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1009, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1081, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 132, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 97, in async_press
    if action == WRITE_MULTI_MODBUS:
       ^^^^^^
UnboundLocalError: cannot access local variable 'action' where it is not associated with a value`

### Integration Version

2025.07.02

### Homeassistant core version

2025.6.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP1ES120

### Connection Method

LSE-3, Modbus RTU over TCP

infradom, 2025-07-18:
I will have a look as soon as I come home this evening

infradom, 2025-07-18:
I was not aware that sofar uses autorepeat buttons, bu I will check again later today. Sorry

infradom, 2025-07-18:
When I made this change, I did verify if it would not have an impact on other plugins. I must have overlooked something. 
I did change the structure of an autorepeat button return value; maybe I made a mistake somewhere. I am verifying the code again and the other plugins. Give me a couple of hours ... (or less hopelully)

flopp999, 2025-07-18:
> Ok found the error in button.py: should be:
> 
> ```
>             if self.button_info.value_function:
>                 res = self.button_info.value_function(BUTTONREPEAT_FIRST, self.button_info, self._hub.data ) # initval = 0 means first manual run
>                 if res:
>                     if self.button_info.autorepeat: # different return value structure for autorepeat value function
>                         reg    = res.get('register', self._register)
>                         data   = res.get('data', None)
>                         action = res.get('action')
>                         if not action: _LOGGER.error(f"autorepeat value function for {self._key} must return dict containing action")
>                         _LOGGER.info(f"writing {self._platform_name} button register {self._register} value {res}")
>                         if action == WRITE_MULTI_MODBUS:
>                             await self._hub.async_write_registers_multi( unit=self._modbus_addr, address=reg, payload=data )
>                     else:
>                         await self._hub.async_write_registers_multi( unit=self._modbus_addr, address=self._register, payload=res )
> ```
> 
> I wiill issue a PR

I tried it but I got an error
```
  File "/config/custom_components/solax_modbus/button.py", line 88, in async_press
    res = self.button_info.value_function(BUTTONREPEAT_FIRST, self.button_info, self._hub.data ) # initval = 0 means first manual run
                                          ^^^^^^^^^^^^^^^^^^
NameError: name 'BUTTONREPEAT_FIRST' is not defined
```

Is this correct?

```
 async def async_press(self) -> None:
        """Write the button value."""
        if self._write_method == WRITE_MULTISINGLE_MODBUS:
            _LOGGER.info(f"writing {self._platform_name} button register {self._register} value {self._command}")
            await self._hub.async_write_registers_single(
                unit=self._modbus_addr, address=self._register, payload=self._command
            )
        elif self._write_method == WRITE_SINGLE_MODBUS:
            _LOGGER.info(f"writing {self._platform_name} button register {self._register} value {self._command}")
            await self._hub.async_write_register(unit=self._modbus_addr, address=self._register, payload=self._command)
        elif self._write_method == WRITE_MULTI_MODBUS:
            if self.button_info.autorepeat:
                duration = self._hub.data.get(self.button_info.autorepeat, 0)
                autorepeat_set(self._hub.data, self.button_info.key, time() + duration - 0.5 )
            if self.button_info.value_function:
                res = self.button_info.value_function(BUTTONREPEAT_FIRST, self.button_info, self._hub.data ) # initval = 0 means first manual run
                if res:
                    if self.button_info.autorepeat: # different return value structure for autorepeat value function
                        reg    = res.get('register', self._register)
                        data   = res.get('data', None)
                        action = res.get('action')
                        if not action: _LOGGER.error(f"autorepeat value function for {self._key} must return dict containing action")
                        _LOGGER.info(f"writing {self._platform_name} button register {self._register} value {res}")
                        if action == WRITE_MULTI_MODBUS:
                            await self._hub.async_write_registers_multi( unit=self._modbus_addr, address=reg, payload=data )
                    else:
                        await self._hub.async_write_registers_multi( unit=self._modbus_addr, address=self._register, payload=res )
```

flopp999, 2025-07-18:
@infradom If i changed to 0 it worked. But then I guess your change may not be a change :)

`res = self.button_info.value_function(0, self.button_info, self._hub.data )`

flopp999, 2025-07-19:
My fault, sorry.
I had removed BUTTONREPEAT_FIRST, for testing when it was not working.

With your new code it works. :)

wills106, 2025-07-19:
Just released [2025.07.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.07.3) with this change.

spetrie13, 2025-07-19:
Similar issue for Solis inverter. Hopefully 2025.57.3 fixes it for solis inverter too.