wills106/homeassistant-solax-modbus #1482: [Bug]: sensor.xxx_pv_power_total unavailable after upgrade

Kars-de-Jong, 2025-07-23:
### Describe the bug

After upgrading to version 2025.07.3, the sensor sensor.solax_x3_mic_pv_power_total no longer shows a value.
I chased this down to this change: https://github.com/wills106/homeassistant-solax-modbus/commit/005b19d22cde13dd10dbb1219532a163033f623c#diff-e856818107a0f316b0b9d266979fb408da134baf630fb126b29688b643fd6123R295.
If I remove the line `datadict.pop("pv_power_total", None)`, it works again.

### Integration Version

2025.07.3

### Homeassistant core version

2025.7.2

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

MU802T

### Inverter firmware versions

ARM 1.38-1.02 DSP 1.33

### Connection Method

USR-N510

### Dongle firmware

N/A

infradom, 2025-07-23:
Ok, thanks and sorry: difficult to solve - I had to introduce this because after another change, pv_power_total doubled with each cycle.
This is because the data dictionary is not cleared anymore with each polling cycle; (consequence of the introduction of scan groups). 
For the time being the solution will break another solution...

I cannot find solax_x3_mic_pv_power_total in plugin_solax.py What  is the entity_key you are using?  There is also a pv_total_power confusing. 

One possible solution would be to check first all the strings, and if there is only one string return that single value ...
I will make a proposal.
Am I right when I assume that the MIC has only one string and the entity is called pv_power_total ? so no pv_power_1 ?

Kars-de-Jong, 2025-07-23:
> I cannot find solax_x3_mic_pv_power_total in plugin_solax.py What is the entity_key you are using? There is also a pv_total_power confusing.

Oh, that's the complete entity-id, it has a prefix. The entity_key is pv_power_total:

```python
    SolaXModbusSensorEntityDescription(
        name="PV Total Power",
        key="pv_total_power",
        value_function=value_function_pv_power_total,
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        entity_registry_enabled_default=False,
        allowedtypes=MIC | GEN | GEN2,
        icon="mdi:solar-power-variant",
    ),
```
 
> One possible solution would be to check first all the strings, and if there is only one string return that single value ... I will make a proposal. Am I right when I assume that the MIC has only one string and the entity is called pv_power_total ? so no pv_power_1 ?

No, it has two strings. pv_power_total is calculated using that `value_function_pv_power_total` function.

Kars-de-Jong, 2025-07-23:
> Could you try with this code:
> 
> ```
> def value_function_pv_power_total(initval, descr, datadict): 
>     single_or_oldsum = datadict.pop("pv_power_total", None)
>     vals = [v for k, v in datadict.items() if k.startswith("pv_power_")] # multi string inverter ?
>     return single_or_oldsum if any(p is None for p in vals) else sum(vals) # if so, return sum, else single value
> ```

No, sorry, that doesn't work. Now it's back to `Unknown`.

infradom, 2025-07-23:
Can you try this code, it should be better since it does not scan the entire datadict dictionary every polling cycle:
``` 

MAX_PVSTRINGS = 10

def value_function_pv_power_total(initval, descr, datadict): 
    # changed: for performance reasons, we should not iterate over the entire datadict every polling cyle (contains hundreds ...)
    total = 0
    i = 1
    while i <= MAX_PVSTRINGS:
        v = datadict.get(f"pv_power_{i}", None)
        if v == None:
            break
        else: total += v 
        i += 1
    return total

```

Kars-de-Jong, 2025-07-23:
> very strange, I was assuming there was only one string without suffix number. I have no clue why the function failed then.
> 
> Is there an entity PV Power 3 ?, or some other entity key starting wih pv_power_ ? If so, is the value None/unknown or zero ? Sorry that is currently is completely unclear to me what happens

No, no more PV power entities.

And please note, I only removed the line `datadict.pop("pv_power_total", None)`, after that it works again.
What is supposed to put it back in the data dictionary?

infradom, 2025-07-23:
my new solution, see last post does not do a pop anymore and is far more efficient