wills106/homeassistant-solax-modbus #1486: [Bug]: Inverter power not reported on X1 Vast

carryonrewardless, 2025-07-24:
### Describe the bug

When I install the Solax Modbus integration (version 2025.07.3) from the UI, I get correct data for Battery 1 Capacity, Battery 1 Power Charge.  I see Inverter Current, Frequency and Voltage but 

don't see an Inverter Power entity at all even though register 2 is correct for this datapoint. 

Consequently House Load calculation returns an incorrect value. House Load Alt does the same. 
Is there any logs or tests I can provide to help with this?

### Integration Version

2025.07.3

### Homeassistant core version

2025.6.1

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

10M080

### Inverter firmware versions

ARM 008.02-000.03   Main DSP	010.02

### Connection Method

WiFi+LAN

### Dongle firmware

Firmware Version 1.006.01

### Detailed Error Log

Unable to reach any logs using above link..

2025-07-24 21:30:54.247 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-07-24 21:30:54.714 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2025-07-24 21:30:59.248 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-07-24 21:30:59.714 DEBUG (MainThread) [custom_components.solax_modbus] device group read done

### Additional context

Thanks for your work on this Solax modbus.
I have a recently install X1 Vast. When am using the Modbus integration with modbus.yaml called from config.yaml most things are working correctly except for the battery capacity (either Total or for Battery 1). I have confirmed using ModbusTcpClient in a python script that the correct input register is 28 as per your plugin_solax.py but the data reported in the HA entity is 5200 when e.g. the battery is at 49%.  I notice that register 35 is also reporting this 5200 number in error.

When I disable the Modbus integration and install the Solax Modbus integration (version 2025.07.3) from the UI, I get correct data for Battery 1 Capacity, Battery 1 Power Charge.  I see Inverter Current, Frequency and Voltage but don't see an Inverter Power entity at all even though register 2 is correct for this datapoint. Consequently House Load calculation returns an incorrect value. House Load Alt does the same.

infradom, 2025-07-24:
I believe this is the same issue as #1482. It should be solved once PR #1478 is merged.
@wills106 : I believe PR 1478 is sufficiently tested for a merge, especially with this issue and issue #1482 that appears on some inverters.

@carryonrewardless are you able to test my code (PR 1482 or development fork https://github.com/infradom/homeassistant-solax-modbus)? or do you prefer to wait until it is merged in the main line ?

Alternatively, it is sufficient to implement the const.py change for this bug:
Replace value_function_pv_power_total by this code:

``` 
MAX_PVSTRINGS = 10
def value_function_pv_power_total(initval, descr, datadict): 
    # changed: for performance reasons, we should not iterate over the entire datadict every polling cyle (contains hundreds ...)
    total = 0
    i = 1
    while i <= MAX_PVSTRINGS:
        v = datadict.get(f"pv_power_{i}", None)
        if v is None:
            break
        else: total += v 
        i += 1
    return total
```

carryonrewardless, 2025-07-24:
Thanks @infradom for the response.
I agree that pv_power_total is an important datapoint to generate but I believe we should also have 
```
solax_inverter_power = solax_inverter_voltage * solax_inverter_current
```
for the AC output.
Can this be added to your pull request or needs to be separate?

carryonrewardless, 2025-07-25:
I implemented the code change for const.py in #2 above and find that [sensor.solax_pv_total_power] returns "0".
I have tested adding GEN6 to line 4307 and am now getting inverter_power data. So this could be added to a pull request or a new one.

infradom, 2025-07-25:
Sorry, I was confusing your request with pv_power total. The const.py update was not relevant (but is needed anyway for some installations)

carryonrewardless, 2025-07-25:
No, I am getting a sensible number for battery capacity percent.

TCWORLD, 2025-07-25:
The battery capacity issue is the same on the Gen5 X1 inverters. I ended up just using "Battery Capacity 1" as that one gives the correct reading.
I think it has something to do with the fact that these inverters can be built in two separate battery stacks, and because the second stack isn't present (I have two batteries, but they are in the same stack), the "Battery Capacity 2" reading is always 0% which seems to confuse the overall battery capacity sensor.

carryonrewardless, 2025-07-25:
In Solax terms @TCWORLD you have one battery but it has 2 modules.

My original comment about battery capacity charge parameter was in comparing the response from the Modbus integration with the response from the SolaxModbus integration. They should nominally give consistent results but the Modbus integration falls short.
My comments were only intended to provide extra context in case that helped with trouble shooting.

wills106, 2025-09-03:
EPS Frequency may return Zero due to my [comment here](https://github.com/wills106/homeassistant-solax-modbus/issues/1486#issuecomment-3232303937)

I can try the alternative register on the X1. But it might be worth getting the wiring correct.
You shouldn't need to manually put the Inverter into Backup mode for EPS to function, I know when Grid connected Backup mode prevents the Inverter from discharging the batteries.

carryonrewardless, 2025-09-04:
No, all EPS values are reporting zeroes.

wills106, 2025-09-04:
> No, all EPS values are reporting zeroes.

Just for clarity, are they all zero when in normal operation mode. Or when in EPS mode and the Grid is disconnected?

carryonrewardless, 2025-09-05:
I did another test and observed all of the EPS values are zero in both Normal mode and in EPS mode when the grid is disconnected.  This is the case for the Solax Cloud app. 
On the inverter LED display the real values show as expected for voltage, current, power and frequency.

wills106, 2025-09-05:
If SolaX Cloud and Modbus are both returning these all as zero then I suspect a Firmware issue.
We have had similar before, where it may work on the Inverter correctly but the Cloud or Modbus doesn't report the values.

carryonrewardless, 2025-09-05:
I'll Query Solax support about firmware version.

closed 2026-10-03 by github-actions[bot]