wills106/homeassistant-solax-modbus #1490: [Bug]: Solax Inverter Modbus 2025.7.4 - no connection / data from my Solax X3 Gen4 Hybrid

Austria777, 2025-07-26:
### Describe the bug

Hi folks, after updating the Solax Inverter Modbus Software to 2025.7.4 , i receive no data/information from my Solax X3 Gen4 Hybrid device. Had to bring back my HomeAssistant Software with Version 2025.7.1 which works fine. As my backup works, i can ensure it's a Software issue with 2025.7.4. 

### Integration Version

2025.7.4

### Homeassistant core version

2025.7.3

### Inverter brand

Solax X3 Gen 4 Hybrid

### Plugin used

plugin_solax.py

### Serial prefix

H34A08I5401184

### Inverter firmware versions

ARM 1.42-1.09   // Main DSP	1.43

### Connection Method

LAN

### Dongle firmware

unknown

infradom, 2025-07-26:
Without logs, it is difficult to find out what is wrong. I am running this version on my Solax X3 Gen4 Hybrid without problems. The HA core version is also of interest to find out what is wrong.

wills106, 2025-07-26:
See #1491 

We need error logs and possibly pyModbus version.

infradom, 2025-07-26:
One more question: how is your itegration configured? The polling times for default, medium and fast may    matter. You can see these timings if you try to reconfigure your integration (you dont need to fully reconfigure, just mention the 3 time intervals): my configuration menu looks like this:

[image]

infradom, 2025-07-26:
Thanks, this log should be usefull. I will start analysing

infradom, 2025-07-26:
first impression is that the Solis error may not be the same as the Solax error. I might be wrong; analysing further ...

infradom, 2025-07-26:
Solax logs would help also, but the logging does not need to be in debug mode; normal mode is sufficient

luftdieb, 2025-07-26:
After switching back to 2025.07.3, everything is working again

infradom, 2025-07-26:
I upgraded my core to 2025.7.3 and run the integraton 2025.7.4 without problems. I did not uninstall the integration. 
@luftdieb and @Austria777 : can you let me know if you uninstalled the integration before you performed the upgrade?

infradom, 2025-07-26:
Regarding the Solis problem with special_settings: 
```
config in an invalid entity category for sensor, as a sensor can not be configured. See documentation. https://developers.home-assistant.io/docs/core/entity/

Changes to enforce validation were added to the sensor base class.

Solution is setting the entity category to diagnostic or leaving the config setting away.
```

So the plugin file for Solis needs to be adapted.
I am unsure however if this error causes the rest of the entities not to be added. Possible, but not sure.

infradom, 2025-07-26:
> > I upgraded my core to 2025.7.3 and run the integraton 2025.7.4 without problems. I did not uninstall the integration. [@luftdieb](https://github.com/luftdieb) and [@Austria777](https://github.com/Austria777) : can you let me know if you uninstalled the integration before you performed the upgrade?
> 
> No, I didn't uninstall the integration before. Just selecting in HACS the previous version and after installation, perform restart of HA. That's all.

One more question: is it normal that your inverter name ends with and _  ( solis_modbus_ ) ?

luftdieb, 2025-07-26:
> One more question: is it normal that your inverter name ends with and _ ( solis_modbus_ ) ?

Can confirm, integration ends with "_". Normally not my natural passion to define '_' character at the end :-)
Do you think that can cause this problem?

All entities don't have this additional character in entity name, like "sensor.solis_modbus_active_power".

infradom, 2025-07-26:
> All entities don't have this additional character in entity name, like "sensor.solis_modbus_active_power".

I find it still strange. I believe this could matter, but I dont understand that version 2025.7.4 would behave differently than 2025.7.3 . The changes are in a different area. I go to sleep now, will continue tomorrow

kelddamsbo, 2025-07-27:
Same problem with Growatt MOD 6000 TL-XH3, no readouts. went back to 2025.7.3, everything ok again

infradom, 2025-07-27:
@luftdieb : In your logs, I see the entry 
``` 
2025-07-26 22:17:32.559 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'host': 'x.y.z.u', 'interface': 'tcp', 'name': 'solis_modbus_', 'plugin': 'solis', 'port': 8899, 'read_dcb': False, 'read_eps': False, 'read_modbus_addr': 1, 'read_pm': False, 'scan_interval': 15, 'tcp_type': 'tcp'}
``` 

I find it somewhat strange that the more recent options scan_interval_medium and scan_interval_fast are not present. The system should use defaults then, but maybe that mechanism does not work. As you reconfigured (renamed) your inverter recently, these entries should now be present. Do you have debug logs for that reconfigured version? Sorry ...

infradom, 2025-07-27:
One more question: maybe the people that encounter problems have disabled the state recorder. This would explain what happens. If tihis would be the case, I could probably find a fix (loosing some added features).

infradom, 2025-07-27:
> Same problem for Solinteg inverter, some (mostly battery related) of the sensors are unknown on 7.4, but worked on 7.3

and again, noone willing to send a logfile or reply my questions . Sorry

michaelbeelman, 2025-07-27:
For what it's worth, I'm running a Solax Gen 1 (SN: MC502TF4172046) and am receiving data since the update to 2025.7.4, but the House Load sensor is null. I've reverted back to 2025.7.3 and now everything is working again.

This image shows the sensor had data until the moment of the upgrade to 7.4. 
[image]

Home Assistant OS 16.0
Core 2025.7.3
Supervisor 2025.07.1
Frontend 20250702.3

Not sure if this is related, or if I should create a new issue for this? I'm currently on holidays and unable to provide logs as I already restored to previous version.

infradom, 2025-07-27:
> For what it's worth, I'm running a Solax Gen 1 (SN: MC502TF4172046) and am receiving data since the update to 2025.7.4, but the House Load sensor is null. I've reverted back to 2025.7.3 and now everything is working again.

Thanks, this may help, but logs is what we need most. Have nice holidays !

michaelbeelman, 2025-07-27:
Understandable, and indeed a special because of the dependency of testing in the wild. As soon as I see the opportunity, I will upgrade again, fetch the de debug logs for troubleshooting and revert back.

Austria777, 2025-07-27:
> > Same problem for Solinteg inverter, some (mostly battery related) of the sensors are unknown on 7.4, but worked on 7.3
> 
> and again, noone willing to send a logfile or reply my questions . Sorry I have searched almost all night on this, but I cannot reproduce this on my system, so we need logfiles and answers:
> 
> * have you disabled recorder/state logging? it should not be disabled for now
> * can you send a logfile
> * have you tried to reconfigure your instance? or even remove it an reinstall it ?
> * what are your configured polling intervals?
> 
> Just saying, yes I also have this problem does not help as long as we do not have logfiles for each inverter brand where the problem occurs.

Hi gents, I'm mo a specialist on HA, but let's try to figure out. Hope this answers will help you:

* have you disabled recorder/state logging? it should not be disabled for now
   would say no, did nothing in the configuration yaml. The only thing i see there is this:
       _http:
  cors_allowed_origins:
    - https://google.com
    - https://www.home-assistant.io
  ip_ban_enabled: true
  login_attempts_threshold: 5
  use_x_forwarded_for: true
  trusted_proxies:
    - 172.30.33.0/24_
 
* can you send a logfile
   right now I'm back on 2025.7.1. To reproduce the error i have to upgrade to 2025.7.4. Which Logfiles would you need and where to find?

* have you tried to reconfigure your instance? or even remove it an reinstall it ?
   No - nothing at all. I only pressed the "update button" in Homeassistant once i got the hint from HA. I updated from 2025.7.3 to 2025.7.4. That's the only thing i did 

* what are your configured polling intervals?
   Medium polling intervall is 15
   Fast polling interval is 15

Hope this helps, and sorry as i said, I'm not a HA specialist, stupid user only ;-)

BR. Werner

Austria777, 2025-07-27:
> [@Austria777](https://github.com/Austria777) Thanks for these answers. Lets hope someone with the same problem can send a logfile. If you want to try yourself, the log files be found under settings > system > logfiles, and then on the top right, there is a download icon.
> 

Where are the logfiles located, what's the name of the logfiles? 
I can upgrade later to 2025.7.4, send you the logfiles and recover my HA from the backup i have, it's not a big deal for me.
I only have to know, which logfiles you would need ;-)

BR.
Werner

infradom, 2025-07-27:
@Austria777: See my answer above for howto obtain the logfile;  even the logfile for your current version would help a bit, since it gives a view on what is different.

luftdieb, 2025-07-27:
Maybe it's not clear how to provide debug logfile. At the Solax integration you can find at the upper right area on the 3 dots and entry Enable debug protocol. Once you have clicked, you should wait a minute and then click again "disable debug protocol" (or similar because my system is not in english).

[image]

You can open the downloaded logfile and check if no senstive data like serial number is included and you can make a search and replace. 
This logfile is really valuable for finding the problem.

Here is my logfile after renaming my integration (without the _ character) with 2025-07-4:
[home-assistant_solax_modbus_2025-07-27T09-43-50.841Z.log](https://github.com/user-attachments/files/21453148/home-assistant_solax_modbus_2025-07-27T09-43-50.841Z.log)

Hope that helps :-)

Thanks for you great support and I hope you will not have too many sleepless nights.

Austria777, 2025-07-27:
@ Luftdieb: THX - got it
here we go ....

[home-assistant_solax_modbus_2025-07-27T09-53-51.029Z.log](https://github.com/user-attachments/files/21453198/home-assistant_solax_modbus_2025-07-27T09-53-51.029Z.log)

infradom, 2025-07-27:
> would this file help you ?

strange log: I see that you are using another solax (http) integration. I assume you use both, but I can hardly see any log lines related to this integration.

Austria777, 2025-07-27:
@luftdieb  THX for your explanation - got it
....here we go .....

[home-assistant_solax_modbus_2025-07-27T09-53-51.029Z.log](https://github.com/user-attachments/files/21453209/home-assistant_solax_modbus_2025-07-27T09-53-51.029Z.log)

infradom, 2025-07-27:
> Here is my logfile after renaming my integration (without the _ character) with 2025-07-4:

Thanks, interisting, your config options still have missing interval_fast and interval_medium enties, this should not be the cause, but we never know.
``` 
2025-07-27 11:42:47.925 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'host': '<removed>', 'interface': 'tcp', 'name': 'solis_modbus_', 'plugin': 'solis', 'port': 8899, 'read_dcb': False, 'read_eps': False, 'read_modbus_addr': 1, 'read_pm': False, 'scan_interval': 15, 'tcp_type': 'tcp'}
```

Removing the integration, restarting HA and reinstalling the integration could give an indication if the error is related to the older setup entries list. If the name does not change, you will not loose state history.

infradom, 2025-07-27:
> [@luftdieb](https://github.com/luftdieb) THX for your explanation - got it
> ....here we go .....

Thanks Werner, same observation in this new log: no relevant solax_modbus 
Except for this line: 

2025-07-27 10:28:37.595 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX-M_eps_frequency is already used by sensor.solax_m_eps_frequency - ignoring sensor.SolaX-M_eps_frequency

Are these two integrations conflicting with each other ?

infradom, 2025-07-27:
@wills106 If you would correct something, I would also like to add some more targeted debug output.

wills106, 2025-07-27:
Looks like the duplicate has been there for a while.
I'll update soon and then can merge in your change as well. Then do a new release.

infradom, 2025-07-27:
Ok, give me some more time to add some debug statements and extra sanity checking code

infradom, 2025-07-27:
> Maybe it's not clear how to provide debug logfile. At the Solax integration you can find at the upper right area on the 3 dots and entry Enable debug protocol. Once you have clicked, you should wait a minute and then click again "disable debug protocol" (or similar because my system is not in english).
> 
> <img alt="Image" width="1382" height="266" src="https://private-user-images.githubusercontent.com/117862420/471192060-b1885632-0160-4251-943d-b88db49771a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM2MTkxODcsIm5iZiI6MTc1MzYxODg4NywicGF0aCI6Ii8xMTc4NjI0MjAvNDcxMTkyMDYwLWIxODg1NjMyLTAxNjAtNDI1MS05NDNkLWI4OGRiNDk3NzFhOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcyN1QxMjIxMjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMTg2NjcxZWY2MjE1MjI0MTFmNTc4ZDFjMWNiODdjNjFlNDUyMzAwNGRkNDdlY2IzNGJkYzljMTIzMWYwZDEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xMiPC7dxnQMEQAw60HStxsfib8njEvnrWygKENxITo4">
> You can open the downloaded logfile and check if no senstive data like serial number is included and you can make a search and replace. This logfile is really valuable for finding the problem.
> 
> Here is my logfile after renaming my integration (without the _ character) with 2025-07-4: [home-assistant_solax_modbus_2025-07-27T09-43-50.841Z.log](https://github.com/user-attachments/files/21453148/home-assistant_solax_modbus_2025-07-27T09-43-50.841Z.log)
> 
> Hope that helps :-)
> 
> Thanks for you great support and I hope you will not have too many sleepless nights.

Thanks @luftdieb, but may I ask @Austria777 or other people to include also a HA restart in the debug output?  

So enable debug output, then do a HA restart (developer tools), wait long enough to be sure that all integrations have properly finished their starutp and then disable the debug output to collect the file. 

The HA startup phase output lines are essential for the error we encounter

wills106, 2025-07-27:
I have just released [2025.07.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.07.5) with the updated debugging and removed the duplicate EPS Frequency.

infradom, 2025-07-27:
@luftdieb : while re-examining your 2 logfiles and the screenshot (for activating debug), I see that the entity keys still contain a double underscore: `solis_modbus__`
Furthermore, the screenshot shows that you have 2 devices in this integratiion (maybe one with the _ and one without ?)
Or do you really have 2 inverters connected?

@wills106 : some of the errors in the log are related to the switch entities (bitmask) whose corresponding sensors have the faulty ...class = CONFIG . Shall I remove these CONFIG class declarations for the sensors? or will you do this?  Of course the switch entities can keep their CONFIG class, only the sensor entities need to be edited.

infradom, 2025-07-27:
> Here is the debug log incl. a fresh reboot of HA.

Thanks a lot, I will dive into it immediately !!

luftdieb, 2025-07-27:
You can disable and enable entities on the setting-device-integration-solax Page in the middle section of the Page. Some of the entities are disabled by default and must be enabled manual by the User. But also some of the entities are Not compatible and cause Overall failure of the Integration. 

> @infradom I am not aware of disabling it. In fact, it seems enabled but unavailable or unknown.
> 
> [image]
> [image]

luftdieb, 2025-07-27:
Regarding my two entries I did not create them by myself and I do have just one inverter connected. I can't delete the entry and I can just disable the Other entty.

michaelbeelman, 2025-07-27:
I disabled measured power and measured power per phase and re-enabled them. Did a reboot of HA and noticed House Load is available again! Not sure what exactly did the trick (likely disabling and re-enabling did it?), as in two distinct attempts it was unavailable, but now it works for me. Please let me know if I can be of any help to further troubleshoot remainder of this issue. Your (round the clock and professional) help is much appreciated.

infradom, 2025-07-27:
> Please let me know if I can be of any help to further troubleshoot remainder of this issue. Your (round the clock and professional) help is much appreciated.

Thanks a lof for your help, I think the experiences of other people may help us find the real cause, but your observation and test may have helped us a lot ...

Have a nice vacation now !

thomas7475, 2025-07-27:
I thing i have the same problem running a Growatt MIC 600TL-X after the update from version 2025.07.03 to 2025.07.05 there a no more updates for the data readouts. I am using pymodbus-3.9.2 i have the debug log attached with the update from 2025.07.03 to 2025.07.05

Installation method Home Assistant Supervised
Core 2025.7.3
Supervisor 2025.07.1
Operating System
Frontend 20250702.3

[home-assistant_2025-07-27T15-59-55.903Z.log](https://github.com/user-attachments/files/21454895/home-assistant_2025-07-27T15-59-55.903Z.log)

luftdieb, 2025-07-27:
After deleting the hub and restart and add newly, integration works fine now with 2025-07-5.
Before the device show 208 entities. Now ot show 158.
Maybe over the 2 years of using some old outdated entities was part of the device which cause compatability issues... 
Thanks for your support !

thomas7475, 2025-07-27:
I can confirm after  disabling the hub and then add it as a new one solved the problem

infradom, 2025-07-27:
PRELIMINARY CONCLUSION for people encountering similar problems (relatively limited number of people have reported this as far as we know yet): 

For the time being, the situation can be fixed by following ssquence:

- removing the integration entirely
- restarting HA
- reinstalling the Integration

thomas7475, 2025-07-27:
i have to correct my self after i add the the hub new it worked for a fer minutes but the a few of the entities stopped reporting and are now reported as Unknown.

[image]
[image]

infradom, 2025-07-27:
@thomas7475  can you send the logs again?

thomas7475, 2025-07-27:
her the lass log i downgraded to version 2025.07.03 everything was working there and than upgraded again to .05 the same entity's stopped reporting. 

[home-assistant_2025-07-27T16-56-55.609Z.log](https://github.com/user-attachments/files/21455124/home-assistant_2025-07-27T16-56-55.609Z.log)

infradom, 2025-07-27:
maybe the new version assigns different identities, what makes both versions incompatible. This does not explain why the entities disappeared in the new version. Did you use long polling cycles ?

infradom, 2025-07-27:
Are you sure they were not disabled by default in the integration? I will check later, have to do other stuff first...

thomas7475, 2025-07-27:
@luftdieb i checked in Solax integration at settings-devices-integration-Solax it is the same you have that a few of the previous available entities are now missing if i revert back to 2025.07.03 they start working normal again .

luftdieb, 2025-07-27:
No, there are not disabled. Just some entities are not working and status just shown as unavailable. 

[image]

infradom, 2025-07-27:
@luftdieb : the missing sensor entities all seem to share  a  

```
  entity_registry_enabled_default=False,
```
declaration.
Their select and number counterpart do not have that declaration and are therefore visible, but without an associated sensor value.

It is unclear why either

-  previous version ignored these declarations.
-  and you enabled them one by one in the past, and this information got lost during the reinstallation (which would be normal for a reinstallation).

Anyway, it doesnt make sense to have the number and selects enabled, and not enable the corresponding sensors.
Before modifying the solax plugin file, I would like to hear the opinion of @wills106 . I will start modifying my version, as preparation for a PR, but I cannot test and @wills106 will have to review/approve.
So the solis plugin needs to be adapted in 2 ways:
- remove the entity_category=EntityCategory.CONFIG declarations for the sensors
- remove the  entity_registry_enabled_default=False, declarations for the sensors that have an enabled number or select counterpart

jsantosdias, 2025-07-27:
Same problem, with X1-Hybrid-5.0kW SK-SU

thomas7475, 2025-07-28:
i am not sure if it helps but found the commit that first introduced the problem it was commit: b949964fa86e4dfdcf40b3cbea3ae1d4f3ac7dab if i checkout any commit before it than it works normal.

infradom, 2025-07-28:
> i am not sure if it helps but found the commit that first introduced the problem it was commit: [b949964](https://github.com/wills106/homeassistant-solax-modbus/commit/b949964fa86e4dfdcf40b3cbea3ae1d4f3ac7dab) if i checkout any commit before it than it works normal.

Thanks, I am fully aware that my commit caused the incompatibility, but as almost noone provides logs, I am in the dark.

thomas7475, 2025-07-28:
her is a new log i first restarted home assistant and than disabled one of the not working entity's grid_frequency waited a moment and than enabled it again.
[home-assistant_2025-07-28T08-53-04.234Z.log](https://github.com/user-attachments/files/21465412/home-assistant_2025-07-28T08-53-04.234Z.log)

infradom, 2025-07-28:
> her is a new log i first ...

Thanks a lot !!!

infradom, 2025-07-28:
I must admit Gemini provided an excellent answer. I still want to use the correct default for new installations/entities

rosenrot00, 2025-07-28:
> I have just released [2025.07.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.07.6)

Seems still not to fix it on my end.

infradom, 2025-07-28:
@thomas7475 : I am now testing with following slightly enhanced version of your code (better default for new entities). The determination of entity_id may not be equal on each system, but it seems to work for me.
``` 
def is_entity_enabled(hass, hubname, descriptor): # Check if the entity is enabled in Home Assistant
    entity_id = f"sensor.{hubname}_{descriptor.key}"
    """Check if an entity is enabled in the entity registry."""
    registry = er.async_get(hass)
    entity_entry = registry.async_get(entity_id)
    # If an entity is not in the registry, it is probably a new one.
    # Treat it as enabled by default.
    if entity_entry is None:
        _LOGGER.debug(f"Entity {entity_id} not found in entity registry, applying default {descriptor.entity_registry_enabled_default}")
        return descriptor.entity_registry_enabled_default
        
    # Otherwise, return the inverse of the 'disabled' attribute
    if entity_entry.disabled:
        _LOGGER.debug(f"Entity {entity_id} is disabled, not adding to read block.")
        return False
    return True
``` 

The 2 calls need to be adapted also:
```
                    d_enabled = d_enabled or d.internal or is_entity_enabled(self._hass, self._name, d) 
...
               d_enabled = descr.internal or is_entity_enabled(self._hass, self._name, descr) 
```

rosenrot00, 2025-07-28:
> A draft PR https://github.com/wills106/homeassistant-solax-modbus/pull/1501 with this code has been published, testers welcome !!!

I made the changes as outlined here. It doesn't solve the issue for me unfortunately.

[26 more replies]