wills106/homeassistant-solax-modbus #1496: [Bug]: 2025.07.5 Sensors working but other elements are not read any longer

rosenrot00, 2025-07-27:
### Describe the bug

While sensors are read like before, options like GMPPT or "Charger Start Time 1" are not shown with the 2025.07.5 version. Reverting back to 2025.07.5 helps.

### Integration Version

2025.07.5

### Homeassistant core version

2025.7.3

### Inverter brand

Solax X3 Hybrid G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A08

### Inverter firmware versions

DSP v1.54 ARM v1.51

### Connection Method

ESPHome component, Solax Pocket WiFi

### Dongle firmware

3.019.01

### Detailed Error Log

As I said, sensor are read as usual, everything else like elements I can control are not shown in the integration.```
2025-07-28 00:12:55.121 ERROR (MainThread) [pymodbus.logging] No response received after 0 retries, continue with next request

2025-07-28 00:12:55.121 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 84 -> Modbus Error: [Input/Output] No response received after 0 retries, continue with next request

2025-07-28 00:12:55.123 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)

Traceback (most recent call last):

  File "/config/custom_components/solax_modbus/__init__.py", line 502, in _refresh

    await self.async_refresh_modbus_data(interval_group, _now)

  File "/config/custom_components/solax_modbus/__init__.py", line 554, in async_refresh_modbus_data

    for group in interval_group.device_groups.values():

                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^

RuntimeError: dictionary changed size during iteration
```

wasinix, 2025-07-28:
Experienced similar behaviour with my Sofar HYD15KTL after upgrading to 2025.07.5.

rosenrot00, 2025-07-28:
If logs help, please let me know. I can provide them.

rosenrot00, 2025-07-28:
2025.07.6 still doesn't work for me. Same result.

[home-assistant_solax_modbus_2025-07-28T10-43-51.610Z.log](https://github.com/user-attachments/files/21467284/home-assistant_solax_modbus_2025-07-28T10-43-51.610Z.log)

infradom, 2025-07-28:
Have you looked at the entity_id's of the entities? it is possible that they have received a _2 suffix at the end. Manual editing and removing the _2 helped for me.

infradom, 2025-07-28:
Are you sure you have opened the config for a failing select, and looked into the entity name to see if it does not contain an extra _2 at the end ? Example: 

[image]

rosenrot00, 2025-07-28:
@infradom I don't have any _2 in my entity names. Basically all entities fail expect for the first five - which potential indicates an a problem reading or parsing them via modbus?

[image]

[image]

infradom, 2025-07-28:
> which potential indicates an a problem reading or parsing them via modbus?

I would expect to see modbus errors in the logs then, all modbus reads seem to succeed, I will check once more if the block that contains a failing entity succeeds. Attertion some debug show decimal registers, while others hexadecimal ...

rosenrot00, 2025-07-28:
You could send me adapted py files with more debug output if needed. I can run them and provide you the debug logs.

infradom, 2025-07-28:
> You could send me adapted py files with more debug output if needed. I can run them and provide you the debug logs.

I am looking at battery heating, and that seems to read well:
```
2025-07-28 19:50:53.312 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xa3 : battery_heating
2025-07-28 19:50:53.312 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xa4 : battery_heating_start_time_1
2025-07-28 19:50:53.313 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xa5 : battery_heating_end_time_1
2025-07-28 19:50:53.313 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xa6 : battery_heating_start_time_2
2025-07-28 19:50:53.314 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xa7 : battery_heating_end_time_2
2025-07-28 19:50:53.315 DEBUG (MainThread) [custom_components.solax_modbus] skipping bytes 20
2025-07-28 19:50:53.315 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xb2 : pgrid_bias
2025-07-28 19:50:53.315 DEBUG (MainThread) [custom_components.solax_modbus] skipping bytes 6
2025-07-28 19:50:53.316 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0xb6 : export_control_user_limit
```

Be aware that the detailed logging is only done for the first  polling cycles, maybe we can increase that number a bit in __init__.py around line 807 (my version):
```
        if self.cyclecount < 5:
            _LOGGER.debug(f"treating register 0x{descr.register:02x} : {descr.key}")
        try:

---

you could also set the  RETRIES = 0 to a slightly higher value  (around line 44) in __init__.py

But I still believe removing the integration entirely (not just reconfiguring) and restarting HA, before a re-installation is worth trying.

rosenrot00, 2025-07-28:
I also found that this sensor is not available anymore since the update. 

[image]

infradom, 2025-07-28:
> As it works when I revert to 2025.07.3 I assume it must have to do with changes that have been made.

yes indeed, the changes were merely related to

- building scangroups, which seems to work fine in the logs
- skipping disabled registers (but your missing data is not skipped according to the logs)

There mlght be a mismatch between the entity_id  of the select/control and the sensor, so that they are not finding each other. This info is not visible in the logs

rosenrot00, 2025-07-28:
> There mlght be a mismatch between the entity_id of the select/control and the sensor, so that they are not finding each other. This info is not visible in the logs

How could that be? Wouldn't I see other entities in this case? I have had 216 entities always. 

[image]

infradom, 2025-07-28:
yes, you are right, grid status is read, register 0x1A according to the logs. How many  entities does it show after the update?

rosenrot00, 2025-07-28:
> yes, you are right, grid status is read, register 0x1A according to the logs. How many entities does it show after the update?

Exactly the same - 216.

infradom, 2025-07-28:
On my system, grid_status is still disabled:

[image]

rosenrot00, 2025-07-28:
> So it does not appear in the list of disabled entities for Solax also ? But it seems enabled in your system.

Sorry if I wasn't clear on this. I enabled grid status one week ago. Since then it was activated on my system. So It is active on my system, however since 2025.07.05 I get unknown for it.

If I disable the grid status entity it moves back to the disabled entities.

It behalves as any other entity.

infradom, 2025-07-28:
maybe we have to add debug output after the treating register message, to show the value it returns

rosenrot00, 2025-07-28:
> maybe we have to add debug output after the treating register message, to show the value it returns

That would help, maybe also if an entity receives an update? My feeling is that the values might be read correctly but the entities are simply not updated.

Honestly, I believe this issue can't be only present on my setup. I guess as the sensors work most users just didn't realize yet.

infradom, 2025-07-28:
around line 853 on my system:
```

        if val == None:  # E.g. if errors have occurred during readout
            return_value = None
        elif type(descr.scale) is dict:  # translate int to string
            return_value = descr.scale.get(val, "Unknown")
        elif callable(descr.scale):  # function to call ?
            return_value = descr.scale(val, descr, data)
        else:  # apply simple numeric scaling and rounding if not a list of words
            try:
     

```

infradom, 2025-07-28:
ok, thanks this is indeed what I wanted to see, 
In  a couple of places, there is a pop() call from the hub.data  (usually self.data) dictonary. This is meant to remove invalid data after a   read failure. Maybe you can explore that path. 

I think I go to sleep now, it has been late the last days ..

infradom, 2025-07-28:
You might be right, my system also suffers from this problem and I did not notice .... The selects are only filled in after I select them manually. That should ring a bell.  Sorry for not understanding the issue you brought up.
I am glad I can reproduce it and will work on it tomorrow

Sorry for the time we lost, but your extra debug info certainly helps

rosenrot00, 2025-07-28:
> Sorry for the time we lost, but your extra debug info certainly helps

No worries, happy to help. I was just not happy to delete the integration as I have helpers that are assigned to the integration. They are not assigned automatically on reinstall. Also I use certain integral sensors that get wrong values when I play around.

I always try to make sure I‘m not the only one suffering from an issue :)

rosenrot00, 2025-07-28:
> Am I right when I say that the problem only applies to select dropdowns, not to number entities ?

I would say on dropdowns values are missing. Slider seem to be okay as well as some numbers, but not all.

[image]

[image]

[image]

rob11-redfordxx, 2025-07-28:
> I would say on dropdowns values are missing. Slider seem to be okay as well as some numbers, but not all.
I confirm the issues with dropdowns.
I did not notice problems with numbers...you have an example?

BTW, how do I revert to version .04 ?

infradom, 2025-07-29:
I think I found the cause: sensor.py and number.py had code that I commented out async_added_to_hass and async_will_remove_from_hass
Uncommenting seems to restore the values

rosenrot00, 2025-07-29:
> Sorry

All good. I‘m glad you found it in that very short amount of time. 

By the way, one can release new versions as release candidate (RC) in Github. People using the integration won‘t receive RC updates automatically but can select them in Hacs for download and install manually. This could make sense on bigger changes. Then you trigger people to test and once it works you can release it regularly. Just as a hint as this very handy sometimes to track down such bugs and test new versions.

infradom, 2025-07-29:
Thanks, my PR is submitted with the fixes (It may already contain your other PR's code as I tested it)

infradom, 2025-07-31:
@wills106 has published a beta release 2025.07.8b1
We would appreciate it if you could test this release as the changes are somewhat related to this bug.