wills106/homeassistant-solax-modbus #1503: [Bug]: Sofar - HYD 20KTL-3PH missing values on text/dropdown boxes

anton4, 2025-07-29:
### Describe the bug

After upgrading to from 2025.06.3 to 2025.07.x integration does not read values for dropdown boxes after HA restart.
Changing values do work, but after HA reboot they all are empty.

[image]

### Integration Version

2025.07.6

### Homeassistant core version

2025.7.4

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP1ES120

### Connection Method

LSE-3 with Modbus RTU over TCP

infradom, 2025-07-29:
Should be the same cause as the other similar issues recently reported.
Fix is on its way with PR #1501

wills106, 2025-07-29:
I have just released [2025.07.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.07.7) with these changes.

marbell90, 2025-07-29:
The Problem is in 2025.07.7 still there.

marbell90, 2025-07-29:
> > The Problem is in 2025.07.7 still there.
> 
> Hard to believe, it looked so much similar to what we experienced. If you want us to investigate further, we will need a debug log that includes a restart of HA
> 
> Have you checked if the select and number entities have not received a new name (with a _2 suffix) ? In that case, renaming them without _2 will solve the problem. you need to open the settings dialog of a failing entity to see this

I had to switch back to 2025.07.3. I attached the Debug-Log. Hope it helps. Thanks.

[home-assistant_solax_modbus_2025-07-29T21-55-50.717Z.log](https://github.com/user-attachments/files/21497886/home-assistant_solax_modbus_2025-07-29T21-55-50.717Z.log)

infradom, 2025-07-30:
> The Problem is in 2025.07.7 still there.
@marbell90 :
Thanks for the log, but I am unsure if your problem is exactly the same as the one posted by @anton4 in the first post of this thread. Which select entities were loaded badly in your system during the period in which you took this log? A couple of entities is sufficient to perform py search.

Are you sure the files select.py and number.py are the most recent ones that came with 2025.07.7 ?

infradom, 2025-07-30:
@marbell90 

Update: we are making progress
If I take the example for feedin_limitation_mode, the sofar file specifies `entity_registry_enabled_default=False` for the sensor part of this entity. The select part of this entity is enabled. This is probably why sofar users still suffer from the problem. 
Previous versions must have ignored this disabled state if the corresponding select was not disabled. I must admit I do not yet understand completely, but finallly I received a logfile that is useful

Thanks
I see two ways of solving this problem:

- we adapt the sofar plugin file (quick fix)
- we adapt the core of the system (__init__.py) so that it behaves like older versions. (I prefer this solution, but it may take longer to solve)

The problem is related to the fact that disabled registers are not scanned anymore, and therefore the sensor part of your entities was not read anymore because of the entity_registry_enabled_default=False statements in the sofar plugin.

Tomorrow, I will start work on the solution for the integration core (second option)
Regards,

Jan

anton4, 2025-07-30:
After upgrading to 2025.07.7 half of the boxes are being populated, but the other half are not.
There isn't errors in logs either about this.

[image]

wasinix, 2025-07-31:
@infradom , I had a quick test with 2025.07.8b1 and it looks identical as 2025.07.7 with your adapted init.py
Compared to official 2025.07.7 it is a progress, as "Energy Storage Mode" is working again, but the dropdowns for "EPS Mode", "FeedIn: Limitation Mode" and "Passive: Timeout" are still empty and not working properly.

infradom, 2025-07-31:
Thanks for the info, The new software should normally search for a corresponding select with the same key, to see if the polling should be really disabled for this entity.

- You could quick fix it by enabling the charger_use_mode sensor (key name) 

- EPS mode is identified with key eps_control . Also that sensor is disabled by default, but again, the system should have recognized that there is an enabled select with the same name
- same for feedin_limitation_mode
- same for Passive: timeout (key passive_mode_timeout)
- 
I have the impression that the entities that fail have a name that is different from their key ( blanks substibuted by _ and converted to lowercase). Anyway, I will try to fix this, as always, a debug log cointaining a HA restart would simplify my work.

In the mean time, the discribed quickfix (enabling the corresponding sensors) would help, but then it would be more difficult for me to see if the issue is fixed in a next attempt to solve this.

anton4, 2025-07-31:
[image]

infradom, 2025-07-31:
Thanks @anton4. On my development fork I have prepared a version in which the failing controls use internal=true instead of entity_registry_enabled_default=False.
I have not yet asked to push this to the @wills106 mainline
In the meantime, I believe you can fix this by enabling the sensors that correspond to these select boxes; See the list of disabled entities for your inverter, click a failing one and enable it, it may take 30 seconds before everything is reloaded.

I continue to search a more flexible solution, but without debug logfiles containing a HA restart, I cannot do much

wasinix, 2025-08-01:
Hi @infradom, thanks for your continued work on this issue!
Please find attached a debug log with HA restart from Solax module 2025.07.8b1
[home-assistant_solax_modbus_2025-08-01T06-54-56.953Z.log](https://github.com/user-attachments/files/21543096/home-assistant_solax_modbus_2025-08-01T06-54-56.953Z.log)

infradom, 2025-08-01:
@wasinix : thanks for the log; this reveals something totally unexpected:

The block of modbus holding register addresses 4106-4178 cannot be read for some unclear reason, all entities in that block therefore become unkwown. I need to ask @wills106 whether this kind of problem has occurred in the past. The error may not be directly related to my changes, but my changes can have subdivided the read operations in different blocks compared to previous version.
Alternatively, previous working versions may have ignored read errors (I may have corrected this bug)

```
2025-08-01 08:54:00.500 DEBUG (MainThread) [custom_components.solax_modbus] ** trying to read holding block 4106 True
2025-08-01 08:54:00.500 DEBUG (MainThread) [custom_components.solax_modbus] Sofar modbus holding block start: 0x100a end: 0x1053  len: 73 
regs: [4106, 4131, 4132, 4137, 4138, 4168, 4173, 4174, 4175, 4178]
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] failed holding block read_error  start 4106 sync_rtc_result ignore_readerror: True
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] failed block analysis started firstignore: True
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] not touching sync_rtc_result 
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping feedin_limitation_mode = None
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping feedin_max_power = None
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping eps_control = None
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping passive_eps_wait_time = None
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping bat_config_charging_voltage = None
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping bat_config_depth_of_discharge = None
2025-08-01 08:54:00.749 DEBUG (MainThread) [custom_components.solax_modbus] popping bat_config_end_of_discharge = None
2025-08-01 08:54:00.750 DEBUG (MainThread) [custom_components.solax_modbus] popping bat_config_capacity = None
2025-08-01 08:54:00.750 DEBUG (MainThread) [custom_components.solax_modbus] popping bat_config_eps_buffer = None
```

anton4, 2025-08-01:
All the select dropdown entities are already enabled, but their statuses are still unknown
[image]

I tried enabling sensor.sofar_eps_control but it's in unknown status and will not change status.
[image]

marbell90, 2025-08-27:
The last Update to 2025.08.4 still didn’t fix the Problem. It’s still the same 3 register which will not update like in [this](https://github.com/wills106/homeassistant-solax-modbus/issues/1503#issuecomment-3140058816) post.

anton4, 2025-08-28:
I upgraded to 2025.08.5 and by some miracle those 2 select entities (EPS mode, Feedin Limitation Mode) started working.
In case you wan't to debug this further i also generated debug log during HA restart.
[home-assistant_solax_modbus_2025-08-28T05-24-49.004Z.log](https://github.com/user-attachments/files/22018112/home-assistant_solax_modbus_2025-08-28T05-24-49.004Z.log)

infradom, 2025-08-28:
Thanks @anton4 : the new log is certainly helpful. 2025.8.5 contains a number of debug output enhancements, but also a couple of corrections. Your problem may have been a side effect of these corrections. The new log will certainly help me understand what happened and why we did not find the root cause earlier.

infradom, 2025-08-30:
> Unfortunately, these three selection fields (EPS Mode, Feedin Limitation Mode, and Solarman Passive: Timeout) still do not work in release 2025.08.5. The corresponding sensors are also no longer provided by the integration. I attached a Debug Log and hope it helps to point out the bug. Thank you for your efforts.
> 
> [home-assistant_solax_modbus_2025-08-28T09-28-44.783Z.log](https://github.com/user-attachments/files/22021916/home-assistant_solax_modbus_2025-08-28T09-28-44.783Z.log)

I have implemented the dependency mechanism that should resolve this. This will be released soon. Thanks for the log and your patience.

closed 2025-08-30 by infradom

infradom, 2025-09-02:
I need a new debug log (with HA restart) and a clear list of the entities that still show problems

marbell90, 2025-09-02:
Unfortunately, these three selection fields (EPS Mode, Feedin Limitation Mode, and Solarman Passive: Timeout) still do not work in release 2025.09.2 The corresponding sensor-entities are also no longer provided by the integration.

infradom, 2025-09-02:
Thanks for the info, I will dig deeper into this, ... hopefully by tomorrow evening

wills106, 2025-09-05:
Just released [2025.09.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.09.7) with additional debugging.

infradom, 2025-09-05:
@marbell90 , @anton4 :
We have released version 2025.9.7.
It will probably not solve your problem, but we added extra debug/info output what will help us understand why failing registers are not removed correctly.
We would appreciate it if you could test this version and send us a debug log that includes a HA restart as desribed in the readthedocs General FAQ 

Thanks for your help during this analysis

infradom, 2025-09-05:
> With Release 2025.09.7 all entities are unknown.
> I attached the Log:

Thanks, analyzing, the logs show a lot of unexpected read errors, very useful.. 
Analyzing ...

wills106, 2025-09-05:
@marbell90 as you are getting a lot of read error how exactly are you connecting to the Inverter?

If you are using an RS485 adaptor do you have the Termination Resistors installed correctly as per the [Sofar-Installation](https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-installation/) guide?

marbell90, 2025-09-05:
Release 2025.09.8 seems to solve the Problem.  Thanks a lot for your Support. 
Great Job.